QuirkyCort/gears #3: Suggestion of languages

martelima, 2020-08-26:
hello developers, how are you?

My name is Mateus, I am a Brazilian robotics teacher, for the first I would like to thank you for the platform, very well elaborated and worked, one of the best simulators I found, secondly I would like to make a brief suggestion, I teach classes for children, and one of their biggest difficulty is the language in the programming blocks, many still do not know a second language because of their age, would it be possible to have some part of the site that can translate the blocks to other languages? a great research point that I could recommend to you is crowdin (https://crowdin.com/), maybe it helps with the development of the platform, but I just have to thank you. It has helped me a lot in classes with children.Thank you.

QuirkyCort, 2020-08-26:
Thanks for the suggestion. I've just started on supporting multiple languages. If you'll like to help, you can take a look at 'public/js/msg.js'. It contains some sample translations, and can be easily edited to add more languages.

DJuego, 2020-09-12:
Hi again!
When I change the language to something other than English (for example Spanish) GearsBot is broken... (It is not possible to access the block interfaces, simulator, menus...) Can the reason for this be that the translations are incomplete? It is strange...

DJuego

QuirkyCort, 2020-09-12:
It's my bad. I made a mistake when refactoring the translation code. Didn't test it thoroughly.

I've already fixed it, so it should work fine now.

DJuego, 2020-09-12:
Great! Thank you! It works! If all goes well you can expect us to send you an updated translation file for Spanish. We have already detected some "errors". :-P I wonder if the texts in the blocks are also translatable....

DJuego

DJuego, 2020-09-17:
This French text box remains even if English is selected.

DJuego

[image]

QuirkyCort, 2020-09-17:
Sorry. A few of the new French text was incorrectly marked as English. I've fixed it.

QuirkyCort, 2020-12-19:
Multiple languages are now supported in most places within Gears. Closing this issue now. Please open a new issue if there are any more problems with translations.

closed 2020-12-19 by QuirkyCort