QuirkyCort/gears #5: Code separation / Importing python modules

humbug99, 2020-09-03:
Last year my FLL team wrote python code using separate modules for basic utility code, mission code, etc, so that multiple team members could be simultaneously working on code for missions in different modules without worrying about code conflicts.

Is there some way to do this with gearsBot?  It appears that currently there is a single python editor, and when I use File->Load Python, it replaces all of the python code in the editor with new code, but maybe I'm missing something.  We need code separation

If this isn't currently supported, can you give some pointers of where to look in the gearsBot code to add this capability?

Thanks!

QuirkyCort, 2020-09-05:
No. This is currently not supported, but it's a good idea.

If you want to take a shot at this, you can look at...

js/pythonPanel.js : You can add an interface here for multiple files (eg. tabs).

js/skulpt.js : Modify builtinRead to handle the reading of files. This is used when your program imports a module.

humbug99, 2020-09-06:
Thanks.  I was looking at Reeborg's World (https://reeborg.ca/reeborg.html), which is a simulator + python with a totally different front end.  That simulator has, in addition to a "Python Code" panel, a single "library" panel which allows you to put code there, then use it by "from library import foo".

I think this will be easier to implement than allowing many code panels + import + load/save, so I will try that first.

humbug99, 2020-10-09:
I had an idea related to this issue that I want to suggest.  I have some team members on my FLL team who are more sophisticated  about coding, and some who are less sophisticated.  I was thinking that the less sophisticated team members could use the Blockly interface as a way of making a first version of their code, which would then turn into python and get modified in python from there by all team members.  They would learn something about python and the ev3 interface in this process.  So far, this can be done with the current GEARS.

However, if other python modules can be imported dynamically, e.g. from the python library pane, then there is no way to call these from Blockly.  It would be cool if either there was some way to drop in a Blockly block that could call arbitrary python code, or, if function definitions in the python library tab could automatically generate Blockly blocks that would be added to the menu, and a Blockly user could drop these into a program.  I think the second way of doing things is less likely to lead to coding mistakes...

QuirkyCort, 2020-10-09:
Been busy lately and haven't had a chance to look through your PR yet. Sorry.

Your suggestion is doable, as long as the functions are in the library only, and not in the main program. You can't edit python code in the main program and still go back to blocks, as there are currently no ways to convert python back into blocks.

Creating blocks from python code programmatically can be done in blockly. Parsing python code to find function definitions isn't hard either. Object oriented code is probably not gonna be a good fit.

humbug99, 2020-10-09:
Point taken about object oriented code.

I was thinking that rather than parsing the entire python text, it could be easier to put a marker on the line before a def for which a block should be created, like this

# Blockly_function
def foo(bar, baz):
    pass

2 possible advantages:  parsing becomes much easier, and also the user doesn't get lots of extra stuff in the blockly menus that may not be wanted.

Does this seem generally useful enough to add, or is it just a weird uncommon use case?

QuirkyCort, 2020-10-10:
Parsing python to find functions is pretty trivial (...line always starts with a "def "), so I don't think adding the annotation is going to make much of a difference there. It can however be useful to have annotations to set options for how the blocks are generated. I would suggest setting the defaults such that annotations are entirely optional; so that a user can get their python code converted to blocks without having to learn the annotations.

You can see https://makecode.com/defining-blocks for an example of how makecode handles it. I would suggest the following annotations format...

Example (with recommended defaults):
`
#% genBlock=True withOutput=True withoutOutput=True color=330 name='' tooltip='' category=""
def foo(bar):
    pass
`

genBlock: Set to false to disable generating a block
withOutput: Set to false to disable generating a block with output
withoutOutput: Set to false to disable generating a block without output
* By default, both will be generated
color: Color of the block.
name: Name of block. If blank, use the function name
tooltip: Tooltip text
category: Which category to put this block in.

QuirkyCort, 2020-12-19:
Code separation PR has been merged. The idea of parsing Python to auto-create blocks is an interesting one, but needs to be fleshed out more before any work can be done there. If you're interested in following up on that, please start a new issue.

closed 2020-12-19 by QuirkyCort