QuirkyCort/gears #10: Problem with Paintball Launcher and the Robot Configurator (Experimental)

DJuego, 2020-09-10:
I'm trying to develop a proof of concept to test the power of the environment. I would also use it to show it to my colleagues.

The idea is to develop a robot capable of disinfecting an area divided into different rooms(similar to maze) for these pandemic times. 
Using the resources already available to GearsBot, the paintball launcher looks like the most natural hydroalcoholic dispenser. :-)

However, I found a problem with the robot design tool. I attach the design. 
It seems that it does not correctly assign the motors to the paintball launcher. 
The magnetic actuator persists even though it has been removed, too.

DJuego

[Detoxy.zip](https://github.com/QuirkyCort/gears/files/5203591/Detoxy.zip)
[image]

QuirkyCort, 2020-09-10:
The attached robot configuration looks fine; the paintball launcher is present and the magnet was removed. I guess you must be looking at the description text. The configurator doesn't update the description, so it remains the same.

I've updated the configurator to auto-generate a description based on the selected components. It should be less confusing now.

QuirkyCort, 2020-09-11:
In your robot, the paintball launcher is attached to the body and not the arm. That's why it is not moving when you move the arm. To fix it...

- Remove the paintball launcher component
- Select the arm
- Add the paintball launcher

This will attach the paintball launcher to the arm instead of the body.

DJuego, 2020-09-11:
...
How to say it? I am genuinely impressed... Close the issue. Please. 

DJuego

closed 2020-09-11 by QuirkyCort