QuirkyCort/gears #6: Feature Request: Soft Reset?

DJuego, 2020-09-10:
I am a regular user of the OpenRoberta tool (and ev3dev python user). I just discovered your project and I'm VERY impressed. Thank you very much for all the effort and time you have invested in a free access tool for online robotics. IMHO some of its features are almost professional. 

I would like to make any suggestions.  For clarity I will make independent post issues for each suggestion.

a) It would be good to have a soft reset where you could select which aspects are reset (the state of the robot, the painting, the walls of the maze, etc.)

DJuego

QuirkyCort, 2020-09-10:
Thanks for the suggestions.

Currently, "Reset" will reset both the robot and the world. If you just want to reset the world, you can first save the robot position using "Save current position to settings" then do a reset. If this is not good enough, let me know your use case and I'll see what I can do.

Btw, if you can, I would appreciate if you can share with me how and where you're using Gears. This is partly so that I can know what to focus on, and partly to convince others to give Gears a try (...I was trying to convince my local WRO organizer to consider Gears for their virtual competition this year to no success...).

DJuego, 2020-09-10:
Thank you! Your clarification seems sufficient for our purposes. :-) And thanks for being so reactive... and sorry about the noise. However, I am pleased that some of the suggestions were already covered. :-). It is clear that I must read all the documentation first. 

We are still in a very preliminary stage of using your project.  Usually the robotics workshops that we give are in person. This will be our first online experience. That is why we have launched a research in the field of online robotics. And so we have come to your project. 

If the team I am part of finally chooses your proposal, we will surely be happy (and honored) to publicize it. 

I can tell you that the sensations are very good and i can't think of any reason why your local WRO organizer ruled out using your tool. That being said; , our initiative has no connection to WRO or any other formal robotics competition. 

DJuego

closed 2020-09-11 by QuirkyCort