QuirkyCort/gears #7: Feature Request: Import/export world scenary

DJuego, 2020-09-10:
It would be really interesting to be able to import and export scenarios as mazes... 

It is important if you work with a class of students to be able to offer an identical simulation scenario. They could download the scenario and load it into the simulator. Maybe it is possible to do it now?

DJuego

QuirkyCort, 2020-09-10:
It's already possible! :-)

When selecting the maze world, set the maze seed. This is just a number (eg. 42). Everyone who uses the same maze seed will get the same maze layout.

You can also save all the settings by clicking "save". This will save a json file to your computer, and you can then share it with your students.

DJuego, 2020-09-10:
Indeed!! Sorry about the noise! 

This also partially solves the other suggestion of the soft reset for the simulator. 

do you consider having a gitter or discord channel for more direct communication and community building?

DJuego

closed 2020-09-11 by QuirkyCort