QuirkyCort/gears #8: Feature Request: Building scenarios

DJuego, 2020-09-10:
It would be extremely interesting to improve the editing capabilities of the scenario.  For example adding the possibility to simply edit walls manually would be a considerable improvement. Perhaps relying on another tool like Blender or Sweet Home 3D... The simulator already supports walls (!!).

DJuego

QuirkyCort, 2020-09-10:
This is possible in the image world. The wiki page describes how to do it; https://github.com/QuirkyCort/gears/wiki/Image-World-Configurations.

This is an example that adds walls and moveable blocks to the WRO 2019 map (...unzip it first, then load the json file)
[WRO 2019.json.zip](https://github.com/QuirkyCort/gears/files/5202015/WRO.2019.json.zip)

This one adds two unmoveable walls (red) and two moveable blocks (green) to an image I found off imgur...
[imageMap_config.json.zip](https://github.com/QuirkyCort/gears/files/5202030/imageMap_config.json.zip)

Unfortunately, it is not all that easy to use right now, as you'll need to edit the json manually.

I'm not keen to support external editors like Blender. Unless you are already experienced in it, it is far too complicated for an educator who just wants to add a few walls. Also, physics engines only support limited shapes, so you can't just draw any arbitrary shapes in Blender and get it to work.

I'll get around to doing up an easy to use graphical world editor when I get some time.

QuirkyCort, 2021-03-17:
There is now a graphical world editor. Look under "Worlds -> World Builder".

You can find some sample files in the "samples -> custom worlds" folder on github. All of the samples can be loaded and fully edited in the world builder.

shravan01, 2021-03-17:
This is great. Its an awesome feature addition also including previous
work. Thanks for all your efforts. We have continued to admire, use and
learn from this.

Warm regards
Shravan

DJuego, 2021-03-17:
I would vote for your project more than once if I could! Thanks for this!

DJuego

closed 2021-03-22 by QuirkyCort