QuirkyCort/gears #9: Feature Quest: Import Worlds in the Arena

DJuego, 2020-09-10:
I haven't tried it yet but I find the concept of the Arena (a multi-agent environment) mind-boggling. :-) Thank you very much for this! It is very promising!

At the moment it seems that it is not possible but it would be interesting to import the scenarios from the simulator into the Arena (for example mazes). It would also be useful to be able to position/orient the robot(s) manually in the Arena before running the programs. 

DJuego

QuirkyCort, 2020-09-10:
Making a single robot world available in the arena is actually trivial, and some of them are already available (ie. Grid world, and image world). The main consideration is that many of them (eg. line following world) is designed for a single robot and doesn't make sense in a multi-robot arena.

That said, it does make sense to have the maze world in the arena, and I've just added it in. You should be able to see it the next time you reload the arena page.

DJuego, 2020-09-10:
Thank you! It's just what I needed. You can close this issue for now! :-D 

DJuego

closed 2020-09-11 by QuirkyCort