QuirkyCort/gears #11: Feature request: Touch sensor

DJuego, 2020-09-17:
I miss the classic and reliable contact/touch sensor.  We work with the Lego Mindstorms EV3 (ev3dev) and it is a common component in our designs.

Surely there is a good reason not to have implemented it yet. Perhaps problems with the physics engine?

DJuego

QuirkyCort, 2020-09-17:
The touchsensor is a low priority as most use cases can be replicated with an ultrasonic or laser sensor (...the latest mindstorm inventor kit no longer includes the touch sensor for similar reasons). The only case where it differs is when you attached parts to the touch sensor to extend its detection area (eg. to build a front bumper).

If you don't need to add parts to the touch sensor, then it's relatively trivial to implement. If you do need to add parts to extend its detection area, then that would be difficult. I can't think of any ways to implement that in the current physics engine.

DJuego, 2020-09-17:
I understand what you are saying. Thank you!

Good. In practice it is an economical, reliable and easy to understand sensor for students. 

In practice the ultrasonic sensor does not work well with obstacles in close proximity (no echo). And the laser is not in the Lego MindStorms kit.

It occurs to me to add a body (hexahedron) that is sensitive to contact (reports collision with a Boolean (0/1) .  It would have customizable dimension and orientation. It would be possible to add to the robot to do of bumper frontal, back, lateral... 

It is true that this component does not represent reality faithfully, but it would be versatile in several ways. And I think it would be useful schematically.

DJuego

DJuego, 2020-09-21:
I can think of another circumstance. We love the idea of running our Gears generated programs on EV3 bricks. Naturally, the fewer changes required the better.

If our design has a contact/touch sensor, how can we generate a compatible program? To begin with, the sensor port should be reserved for that sensor. We don't have this problem with OpenRoberta...

DJuego

QuirkyCort, 2020-12-20:
Added touch sensor. It has configurable width and depth, so you can make a wide bumper with it. Unlike the ultrasonic sensor which uses raycast to detect objects, this one uses physics collisions to detect touch. This makes it better at detecting contacts with small objects.

DJuego, 2020-12-20:
Well! @QuirkyCort . This was a premature and absolutely unexpected Christmas present. I am quite grateful to you. [I have already tested it](https://mega.nz/file/Ns4CiDJL#hJKc8ICK179shlm1RGW8VZzja2XxN7D1OSGQ3NeaKVQ) and I am absolutely delighted. The behavior looks perfect! I sincerely missed it. And it's kawaii too! 💯 :-)

Again, thank you very much!
This issue has been resolved VERY satisfactorily. You can close it. 👍 

PS:
An interesting aspect is the small delay (in this case 10 milliseconds) that needs to be added in the loop. If not, the program does not work. Indeed; in real programs it also becomes necessary sometimes (I imagine that to limit the rate of queries to the sensors), but I had not detected this need before in Gears (I think).

QuirkyCort, 2020-12-20:
Every sensor needs a small delay to prevent locking up the browser tab. I've added them into all the other sensors, but missed it out when I was implementing the touch sensor. I've just added it in, so you won't need to add the delay manually any more.

For a real robot, there's no need for the delay, either because the OS is multi-threaded, or because the CPU is only controlling the robot and doesn't need to do anything else. For a simulator running in the browser, the same thread that runs the robot program is also responsible for rendering, detecting mouse clicks, 3D simulation, etc. So without a delay, nothing else can be done and the tab locks up.

closed 2020-12-20 by QuirkyCort