QuirkyCort/gears #12: Bug?: Paint does NOT disappear...

DJuego, 2020-09-17:
About the paint: ttl: "Time-To-Live in milliseconds. After this duration, the paintball will be removed." I test with 3000, 500, 3...

Changing this value seems to have **no** effect (for example in Grid Map). Does it also depend on the world you are in as it seems to be the case with the amount of ammunition?

DJuego

QuirkyCort, 2020-09-17:
The Time-To-Live removes only the paintball, not the paintball splatter. The splatter sticks around until the world is reset.

This parameter is there to prevent a paintball that falls off the side of the map from falling forever (...and continuing to put a load on the physics engine).

Do you have a use case that requires the paintball splatter to be removed after some time?

DJuego, 2020-09-17:
Oops. I see. I don't think I would have guessed that nuance. :-)

Well. It just seemed to me a natural parameter (it happens with real painting if you take a long enough period of time :-D), although I understand that its implementation is not trivial. I guess I can live without it for now anyway...

 Going back to my example of disinfecting robots, time progressively weakens the effect of the disinfectant in the environment, so it has to be reinforced periodically. 

DJuego

DJuego, 2020-09-17:
I can think of another circumstance. We love the idea of running our Gears generated programs on EV3 bricks. Naturally, the fewer changes required the better. 

If our design has a contact/touch sensor, how can we generate a compatible program? To begin with, the sensor port should be reserved for that sensor. We don't have this problem with OpenRoberta...

DJuego

QuirkyCort, 2020-09-24:
Not hard to do, but I've been busy lately...

I've added in a new parameter for the paintball launcher "splatterTTL". By default, it is set to "-1" (...last forever), but you can set it to any value (...milliseconds) you want using the robot configurator. The site is already updated, so you can try it out whenever you want.

DJuego, 2020-09-24:
Perhaps I should clarify that I am referring to a local Time-To-Live. 

That is, each splash would have its own internal timer. That's why I suggested that the implementation of this feature might not be trivial. 
Maybe you implemented a global timer that eliminates all splashes in the scenario? It's not what I was looking for originally but I can manage it. :-}

In any case it doesn't seem to work well at the moment...

Cheers!

[image]

[Desifeccion.zip](https://github.com/QuirkyCort/gears/files/5278667/Desifeccion.zip)

QuirkyCort, 2020-09-25:
Sorry. Didn't test it properly. Should work fine now.

DJuego, 2020-09-25:
:-O Amazing! Thank you very much for this! Thank you very much for your support!  The behavior is exactly what I requested.
You can close this issue and score yourself a great success.  :-)

DJuego

closed 2020-09-26 by QuirkyCort