QuirkyCort/gears #18: Feature request: "Import zip package from your computer"

DJuego, 2020-10-06:
Hello again! :-)

Is there any reason why there is no reverse function to "Export zip package to your computer"?

NOTE: The expected result of the operation is the automatic loading of the block code/python code AND the robot design associated with these codes.

DJuego

QuirkyCort, 2020-10-08:
No reasons really. The zip package export was created for the Arena, where it is too much of a hassle to configure each robot one-by-one.

There are no reasons why I can't have a zip package import in the single robot Gears, but I just never added it.

DJuego, 2020-10-08:
I see. I think the zip package export/import  introduces  in **Gears** a primitive version of the concept of projects, which I find really useful and comfortable.

In that case I formally and humbly request that you include this feature in your list of features to do. You decide the priority. :-)

Thank you for being there.

DJuego

humbug99, 2020-10-09:
I think this is a good idea too.  I was thinking about adding it with the code separation pull request, but decided to finish that PR and add it later.

Canx, 2023-12-15:
Hi,

I'm also interested in this feature for my class. @humbug99, did you make any progress in that PR? Maybe I can help, but first I have to be familiar with the code...

QuirkyCort, 2023-12-21:
@Canx Can you share how you would use such a feature in your class?

For my classes, I often use the URL generator to create a custom link that automatically loads a robot. I suspect there is a good chance that this will meet your needs. You can find the URL generator under "Help => URL Generator".

Canx, 2023-12-22:
Hi @QuirkyCort! I haven't used URL Generator, I can see that it can load robots and worlds and restrict the available blocks (very interesting for progressive learning!).

I see URL Generator for a more guided task where as a teacher I can restrict options available, which is great for starting the basics. Sure I will use it!

For this I was thinking on a more open task where the student can select or design their own robot and world, and then can upload their project (zip file) to our Moodle course to be graded manually.

PD: BTW my students are loving the challenge world (using the query worldScripts=challenge) 

PD2: I just seen you added worldScripts=challenges_basic, these gems whould not be so hidden! :heart_eyes:

QuirkyCort, 2023-12-22:
The extra worlds (..."challenge" and "challenges_basic") are for a couple of my classes. Everyone is welcome to use them, but I didn't include them into Gears by default as they are frequently modified. I try to avoid making breaking changes to any of the default worlds, as other educators may rely on them for their classes.

The lessons associated with those two extra worlds can be found here and are freely available for everyone to use...

"challenge"
https://aposteriori-sg.github.io/tutorials/17-Robotics-with-Gearsbot/10-Introduction/10-Course-Overview.html

"challenges_basic"
https://tutorials.aposteriori.com.sg/17-Robotics-with-Gearsbot/10-Introduction/10-Course-Overview.html

Both are designed for younger students and are targeted more at computational thinking rather than robotics.

The original issue in this thread relates to packaging the code and the robot together. You can already export such a zip file, but the exported file can only be loaded in the multi-robot arena. You can however, extract out the json/py/xml files from the zip package and load them individually for grading. Would this meet your needs?

The robot and world are created separately using the Robot Configurator and the World Builder, and would need to be exported to their individual json files. Would it suffice to ask the students to upload 3 files into Moodle? One for robot, one for world, and one for code?

Canx, 2023-12-23:
Hi @QuirkyCort, in the short term I'll extract the zip file and proceed as you said. For my students (middle school) is a little more complexity, specially if they want to continue working at home. They come from a Scratch background and they were used to it, but we will survive!

Thanks a lot for sharing your lessons and in general for creating Gearsbot!

PR #139 (2025-05-21): Option to load zip file from computer
Adding option to load zip file from computer; to resolve #18 .
Tested with two simple exported zip files, on Firefox, Chrome & Edge.

Note:
- Doesn't load additional python modules, beyond the main script.
- Warns user if loaded python script is modified WRT loaded blocks, by raising the "python cannot be converted back into blocks!" message. Maybe deserves a dedicated warning message?
closed 2025-05-21 by QuirkyCort