QuirkyCort/gears #19: Feature request: Messages

DJuego, 2020-10-11:
Surely this is already being considered. And I understand that it is a low priority and it has important complexities. However, allow me to open an issue for the record. :-) 

Since there is the arena I think it would be very interesting to have blocks for communications between robots (Open Roberta has them). It would be possible then to address the multiagent robotics with **Gears**.

DJuego

QuirkyCort, 2020-10-11:
This isn't hard to implement, but deciding on the API is a little more frustrating. I can follow Pybricks or roll my own...

Pybricks
This is similar to the EV3-G message blocks, but requires one brick to identify itself as the server. I can generate Pybricks code that can work with a real EV3, but it won't work on ev3dev.

Roll my own
This allows me to dispense with the need to specify a server / client. It'll also allow me to send more advanced data types (eg. list, dict, key/value pairs), and to send to a channel (eg. team channel, everyone channel) and not just an individual robot. But the generated python code won't work on any EV3.

In both cases, the python code won't work with ev3dev.

What do you think? A better API that doesn't work with real robots? Or a poorer one that has better compatibility?

DJuego, 2020-10-11:
Good. We work with ev3dev. :-P

I wonder if it would be possible to implement a Python library for ev3dev that would encapsulate all the complexity of the communication (Bluetooth and/or Wifi). Also with the ability to serialize/deserialize advanced data types.

 The code generated by Gears could make use of this library.  

It would be the user's responsibility to have pre-installed this library in the ev3dev brick.

DJuego

QuirkyCort, 2020-12-20:
Added radio messaging in the lastest commit.

I'm taking the roll my own route. It makes for a simple and capable API, but won't work on a real robot.

Creating a messaging library for ev3dev isn't something that I'll like to get into. Bluetooth connections are too much of a pain to deal with.

DJuego, 2020-12-20:
Well @QuirkyCort . Another unexpected Christmas gift!!

My **deepest** thanks for this very valuable feature for multi-agent robotics.  I just made my debut at the Arena with this.  :-)
The only thing I have missed is a wiki associated documentation, but I fully understand that it will take time. 
I have some questions such as how **teammates** are defined. (?)

DJuego

PS:

I attached my first test! It works! :-)
I only had one problem uploading the files to the Arena. It seems that when you upload a file  of the same name for updating it doesn't update. 
For instance;
1)
Choose File => Select Sender_Robot1.zip
Choose File => Select Sender_Robot1.zip Does not update.
2)
Choose File => Select Sender_Robot1.zip
Choose File => [CANCEL]
Choose File => Select Sender_Robot1.zip Update successful.

[Receiver_Robot0.zip](https://github.com/QuirkyCort/gears/files/5721372/Receiver_Robot0.zip)
[Sender_Robot1.zip](https://github.com/QuirkyCort/gears/files/5721373/Sender_Robot1.zip)

QuirkyCort, 2020-12-24:
Player 0 and 1 are in one team, while player 2 and 3 are in the other. This is currently hardcoded and cannot be changed.

Seems that the "onchange" event doesn't trigger when the new file has the same name. Shouldn't be hard to fix, but I'll need to make some changes to the UI to make it neat.

QuirkyCort, 2020-12-27:
Fixed the problem in the arena. Now when you upload files with the same filename, it should work correctly.

In the process, the appearance of the file input was also improved, and you can now drag and drop files anywhere within the player frame.

DJuego, 2020-12-27:
Thank you for offering such an outstanding feature as communication between robots... and thank you also for taking care of the small details. Now uploading files to the arena seems much more comfortable. :-)

With your permission I close this issue resolved so brilliantly. 

DJuego

closed 2020-12-27 by DJuego