QuirkyCort/gears #23: Move Steering with Direction block appears to not work correctly

stemrobotics, 2020-10-20:
If  you put a Move Steering with Direction block in a loop, with a fixed direction value, the robot should move in a circular motion. The direction value should control the diameter of the circle traveled. This works for direction values of 15 or greater but for values below 15 the motion may or may not be circular and if circular, the diameter is not what you might expect. If 15 causes the robot to travel in a circle, you would expect 10 to also travel in a circle, though with a larger diameter. In my case, a value of 10 results in the robot travelling in a more less straight line at angle 10. I discovered this trying to drive a straight line using gyro values fed into the Move Steering with Direction block.

humbug99, 2020-10-21:
Interesting.  I see something like this at speeds above 80.  For now, I suggest you try running the robot at a lower speed.  I see some funky behavior similar to what you describe at speed 100, but if I turn the speed down to 50 or 75, then using just a single Move Steering with Direction (blockly) block for 20 seconds does what I would expect for both low and high values of direction and low and high speeds, that is, it drives in bigger and smaller arcs with different values for direction.  

I don't think this is related to acceleration, because stepping up the speed gradually still gives me strange behavior around speed 90 and higher.  See the attached python code and screenshot with robot trace.  The radius of curvature of the robot is pretty consistent up until speed 80 (blue curve), but at speed 90 (green curve) the radius is bigger, and at speed 100 (red curve) the robot goes almost straight. 

The robot trace was made using a custom simulator running the code from Pull Request 17.  The python code will still run on the standard gears simulator as long as you set USE_PEN = False at line 30.

[image]

[pen_move_steering.py.txt](https://github.com/QuirkyCort/gears/files/5415382/pen_move_steering.py.txt)

stemrobotics, 2020-10-21:
Thanks for the reply. Yes it appears speed is the issue. At high speed direction below ~15 does not track as expected but if you lower speed then the tracking becomes as expected. I think if I remember right, the Move Steering with Direction is actually implemented as left/right motor commands with the direction component added/subtracted from the left/right speeds. Hopefully this is a correct memory as I just started reading the GB source yesterday. If that is the case, using 100% power would not leave any room for the steering offset. I just ran some tests that bear out this explanation.

humbug99, 2020-10-21:
I think that this is a mismatch between the motor maximum speed, defined at the python layer as 1050 [motor.py, line 218] (in tacho counts), and in the javascript layer as 800 (also in tacho counts, Wheel.js, line 11).  The python layer calculates wheel speeds based on its maximum, but at high speed one or both of these get limited by the javascript layer, leading to a difference in wheel speeds that doesn't match the input steering amount.

Maybe this could be fixed by having the python motor get its max speed from its wheel, but it would be good to know whether 1050 or 800 is the correct maximum.

stemrobotics, 2020-10-22:
Pardon my ignorance, but how do you use the python code above with the pen simulator? I'm new to python and js so I'm don't understand how you ran that test code in the simulator. 

I have the pen simulator downloaded and running but I don't see the pen as a robot component or the pen-line-follower as a sample robot as the source code suggests I should.

humbug99, 2020-10-22:
I'm not sure where you're stuck, so the following is maybe overly detailed.

Until the pull request gets merged, if you want to run modified code, you need to run the simulator on your own computer.  See the Install section on the main readme.  Basically:
-- If you don't have python installed, install one of the Anaconda packages and create a python3 environment.  The Anaconda getting started instructions are good.
-- clone or download the git repository you want to run (this one, or one from a pull request)
-- activate your conda environment and cd to the top level of the repository
-- type "python -m http.server 1337"
-- point your web browser at http://localhost:1337/
At this point you should see the gears interface like usual, but if you change any of the javascript code or the python code, the simulator will do something different.  If you use the Pull Request 17 (aka pen2) simulator code, you can upload my python script from above and run it with the default robot, and you should see the pen trace.  If you run the python script above against the regular simulator, you should get an error at line 32.

Sample python code for using the pen and robot with pen are in the /samples/python and /samples/robots directory respectively.
The sample python code can be used with the default robot.

Another way to see if the pen2 code is running on your computer is to open the blockly tab, using the default robot.  You should see a Pen section with 3 blocks (Add pen, start drawing, stop drawing).  You can test it by including the add Pen and start Drawing blocks along with a few simple movement commands, and you should see a pen trace.

At the moment, you won't see the Pen in the robot configurator, nor will you see the robot-with-pen in the Select Robot interface, but you can use Robot/Load from File to upload the sample pen robot.  I'm holding off on documenting the pen or fully implementing the blockly pen interface until I get some feedback as to whether the PR is likely to get merged.
For now you can figure out the pen interface by either looking in /public/ev3dev2/pen.py or by looking at the sample code.
If something about the pen doesn't work and/or there is an error message, add a comment to PR 17.  

If that doesn't get you unstuck, please be a little more detailed about what's going on and I'll try to help.

stemrobotics, 2020-10-22:
Thanks for that detailed explanation. I have it working. One thing, using the pen block in my own sample code that drives in a circle, the color of the pen starts blue with a nice width then narrows out of sight at about 180 degrees ten starts widening up and turns black as the robot completes the circle. As the robot traces circles this blue to black wide to thin trace repeats. If I drive straight the line keeps the blue color and width for the length of the drive. I would assume the pen block would be fixed color and width since you have no ability to select these attributes on the block.

Another question, which I have probably just missed...how do you set the origin point of the robot in the simulation space to something other than the center?

humbug99, 2020-10-22:
Yes, this happens on many curved paths.  I'm not entirely certain why, as I'm new to using the babylonJS library that does both the visualization and the physics.  My theory is that the color change happens because the line flips over (the two sides of the line change which is inside and which is outside).  I haven't worked out how to fix it yet.

You can change both the width and and color from python if you want to. 

Changing the robot start point - I think it is in the World control panel, near the bottom.

humbug99, 2020-10-22:
The strange appearance of the trace path turned out to be a bug.  There is an update in the pen2 PR branch.  Thanks for motivating me to fix it.

stemrobotics, 2020-10-22:
That looks good.

QuirkyCort, 2020-10-24:
Hi all. Thanks for the investigation into this issue.

The mismatch between the max speed defined in Python and Javascript is **deliberate**. The Python library is based on ev3dev, which uses 1050 tacho counts per second as the maximum for a large motor. In practice, this speed is unachievable by a real robot. The max speed of a real robot will depend on the load and battery, and will typically be around 800 - 900 tacho counts per second. In Gears, I choose to limit the simulated motors to 800 tacho counts per second in the Javascript code.

If you're using the EV3-G software, this effect remains present, but is less distinct, as EV3-G choose a lower max speed constant. But in both ev3dev and EV3-G, you typically will not see a difference between 95% and 100% speed (...might be able to see it using fresh batteries and running with no load).

When it comes to turning, EV3-G will turn correctly even at high speed, as it synchronises the two motors (...if one motor doesn't turn enough, it'll slow or stop the other motor). Ev3dev does not synchronise the motor, so turns tends to be incorrect at high speed.

So what you see as a bug in Gears here, is simply a result of Gears mirroring the shortcomings in a the real robot. It forces the user to consider these issues (eg. motors don't always run at the requested speed), and create an opportunity for them to explore different ways of dealing with it (...slowing down is one way, another way is to implement a control loop that takes the motors' tacho count as feedback).

closed 2020-11-07 by QuirkyCort