QuirkyCort/gears #26: it is no longer possible 'to overlap' hexahedrons in the worlds? (Oct 28 2020)

DJuego, 2020-10-28:
Sadly, it seems that with the last commit (Oct 28 2020) it is no longer possible to overlap hexahedrons in the worlds. Is this a intentional "fix"? This (anomalous) feature is useful for lack of more polyhedron types in Gears. :-(

DJuego
[image]
[image]

QuirkyCort, 2020-10-29:
You must be using the github version! A recent PR changed the units for rotation into degrees, breaking compatibility with previously created configuration files. I've just reverted it back to radians, so it should be fine now.

DJuego, 2020-10-30:
Yes, yes. I was using  the github version! Sorry! The problem has been resolved with the last commit! Thanks again, @QuirkyCort  I close the issue!

DJuego

closed 2020-10-30 by DJuego