QuirkyCort/gears #28: Questions about custom worlds

stemrobotics, 2020-11-11:
I  am trying to understand the customization of worlds and am struggling to figure out how it all works. I will start with one question and see where it leads: 

If I save the Grid world to a json file and add obstacles or magnetic objects, then load it, the rendered world does not have the obstacles or magnetic objects. They only seem to work when the wordName is "image". Seems like adding obstacles or magnetic objects to the simple grid world, or any of the predefined worlds would be desirable.

QuirkyCort, 2020-11-11:
Right now, every world is fully independent of each other, and only the "image" world supports obstacles and magnetic objects.

In the long term, I expect that the "image" world will be refactored into a base world that the others inherits from, but that's not a priority now. That's something to be considered only when the image world's capabilities are firmed up, and the world editor will likely come before that.

Tenzors, 2021-01-01:
Any recommended simple 3D software for building objects and export to json?

macjones13, 2021-01-04:
I teach robotics and created an image world for my students to use, but when they trying to load the JSON file for it , they can't get it to show up (the robot looks like it's tumbling through space).  Is there a something I'm doing wrong in creating the world or sharing it?  Thanks.

DJuego, 2021-01-04:
Hi! 

I recommend that you share the image world and JSON so that we can try to reproduce the problem and make a diagnosis...

DJuego

macjones13, 2021-01-04:
Here is the JSON:

{
  "worldName": "image",
  "options": {
    "image": "",
    "imageURL": "https://photos.google.com/photo/AF1QipPPwEyLzIwrdRzKcTj1eq_sf5JnQbubzBIkaN5Y",
    "length": 100,
    "width": 100,
    "imageScale": ".35",
    "wall": true,
    "missions": true,
    "wallHeight": 8.8,
    "wallThickness": 3.7,
    "groundFriction": 1,
    "wallFriction": 0.1,
    "groundRestitution": 0,
    "wallRestitution": 0.1,
    "obstacles": [],
    "magnetics": [],
    "objects": [
    {
      "type": "box",
      "position": [56.55,-30.1,4.4],
      "size": [3.7,45.72,8.8],
      "rotation": [0,0,0],
      "color": "#000000FF",
      "physicsOptions": "fixed"
    },
    {
      "type": "box",
      "position": [56.55,45.4,4.4],
      "size": [3.7,15.2,8.8],
      "rotation": [0,0,0],
      "color": "#000000FF",
      "physicsOptions": "fixed"
    },
    {
      "type": "box",
      "position": [22.6,22.6,4.4],
      "size": [3.7,60.86,8.8],
      "rotation": [0,0,0],
      "color": "#000000FF",
      "physicsOptions": "fixed"
    },
    {
      "type": "box",
      "position": [-11.4,-15.4,4.4],
      "size": [3.7,76,8.8],
      "rotation": [0,0,0],
      "color": "#000000FF",
      "physicsOptions": "fixed"
    },
    {
      "type": "box",
      "position": [-56.6,4,4.4],
      "size": [3.7,30.5,8.8],
      "rotation": [0,0,0],
      "color": "#000000FF",
      "physicsOptions": "fixed"
    },
    {
      "type": "box",
      "position": [-35.8,20.8,4.4],
      "size": [45.2,3.7,8.8],
      "rotation": [0,0,0],
      "color": "#000000FF",
      "physicsOptions": "fixed"
    }
    ],
    "startPos": "center",
    "startPosXY": "",
    "startRot": "",
    "imageFile": "blob:https://gears.aposteriori.com.sg/062f30c4-6242-4e15-913b-e1fbc0bda2b4"
  }
}

DJuego, 2021-01-04:
I have to admit that I do not use web links to share the image of the world. In my case students launch a local web server with Gears. Then I attach the JSON file and the world image file.

Anyway I have checked that the JSON is well formed but the links don't work for me. Also, it seems that you need a gmail account to access the resources... Maybe some other folks can tell you more...

DJuego

QuirkyCort, 2021-01-04:
A few problems...

1) That is a URL for a webpage, not an image. You'll need to get the image URL by right clicking on the image and selecting "Copy Image URL".

2) Only some image hosting sites will work and Google photos is not one of them. See https://webglfundamentals.org/webgl/lessons/webgl-cors-permission.html for details. I recommend using Imgur instead (...free and don't even need to sign-up an account).

3) There's an "imageFile" line near the bottom. That should be removed entirely. When removing that line, you'll also need to remove the last comma on the "startRot" line. That option appears if you load an image from your computer and it will not work when the configuration file is shared with others.

macjones13, 2021-01-05:
I got it fixed, thank you.

QuirkyCort, 2021-03-12:
The original question is about adding obstacles to the grid world. This is now possible (...using the newer "objects" format) as grid world has been refactored to use a base world with object support. Other worlds that uses the base world as a parent will also inherit such capabilities, but currently only the grid world and the test world do so.

Some added capabilities includes the ability to create a cylindrical ground and tile the ground image. This is just the first steps towards a graphical world editor and there will be more enhancements coming.

QuirkyCort, 2021-03-22:
With the introduction of the World Builder, all of the issues here should be resolved.

closed 2021-03-22 by QuirkyCort