QuirkyCort/gears #30: Issue with .svg files in Ubuntu-20.10

DJuego, 2020-11-16:
Good! This software is incredible. It exceeds my expectations! Thank you for this!!

In Windows 10 everything seems to work great. 

Now I'm trying it on the Ubuntu 10.12 Linux distribution. 

Everything seems to work fine until I try to load the scenario. It doesn't load the image! It feels like it's because the .svg file is "too" big. In Windows it works perfectly. 

Attached two svg files: **good**.zip works. **bad**.zip does not work. :-(
You can compare it with _Generate from image World_

Any hints?  I'm working with current git master version. 

Thanks in advance!

DJuego

[bad.zip](https://github.com/QuirkyCort/gears/files/5548607/bad.zip)
[good.zip](https://github.com/QuirkyCort/gears/files/5548608/good.zip)

DJuego, 2020-11-16:
Well. I think several points need to be clarified:

1. I am working on a virtual machine on Windows 10.
2. I have tested Mozilla Firefox and Chromium with identical results.
3. I have tried with the web https://gears.aposteriori.com.sg/. Same result. :-O
4. I tried to load a .png file exported from the original .svg. Same result! :-OO The .png looks correct. I have attached it.

I can think of several explanations.
A problem with the handling of the video memory or similar of the virtual machine?
A problem with loading  big images in **Gear** when working on Linux Ubuntu 2020.10?
...?

DJuego

[png_bad.zip](https://github.com/QuirkyCort/gears/files/5549201/png_bad.zip)

QuirkyCort, 2020-11-16:
It's a texture size issue. Different GPU (...and maybe driver?) will have different limits.

One solution is to reduce the texture size (eg. half it) then set the scale factor (eg. scale factor 2) to restore the correct size. This will reduce the texture resolution, but retain the correct world size.

DJuego, 2020-11-24:
Sorry for my very delayed answer! :-/

It is clear that you are right. I was trying to solve the problem in my virtual machine but not without success. What I have found is that Gears works well under Ubuntu installed on computers. :-)

It's certainly not a problem that has to do with Gears, but it's good to know. I'm happy to close the issue and thank you for your quick response!

DJuego

closed 2020-11-24 by DJuego