QuirkyCort/gears #33: Feature Request: Football/Soccer Game

sgmurray, 2020-12-11:
I am interested in making a football/soccer inspired game for gears. How does the community feel about what I have done so far and my plans for future work? 

What I have on my local copy so far:

I added a "football sensor" which tells the robot the position of a magnetic ball. A robot with a forward facing magnetic actuator can grab and kick a magnetic ball.  You can see a single robot tracking a ball and scoring a goal in the Youtube video below. 

[https://youtu.be/tplk9TrJF-w](https://youtu.be/tplk9TrJF-w)

Plans for future work:
1. Create an arena game mode for 2 v 2 table soccer inspired game. 
2. Allow the "football sensor"  to provide the location of other robots in addition to the location of the ball. 

The gameplay of the proposed arena game mode is described in more detail below:

GEARS Football is a table football/table soccer/foosball inspired game. The goal is to simplify the game of football/soccer such that it can be played by robots built and programmed in GEARS.

GEARS Football is played by 2 teams (red vs blue) of 2 robots (attacker and goalie) each for a total of 4 robots. The rectangular field is divided into 4 protected rectangular zones. Each robot is assigned a protected zone which it is not allowed to leave for the duration of the match. 

The protected zones are separated by “invisible fences”, which prevent one robot from entering another robot’s protected zone. There is a gap under the invisible fence which allows the ball to roll under the fence and pass between zones. 

The ball is magnetic and can be grabbed and kicked using the magnetic actuator. Robots use a virtual (this doesn’t exist in real life) “football sensor” to get the position of the ball and the position of the other robots on the field. 

A shot clock is used to keep the game moving. Once the ball enters a robot’s protected zone, it has a time limit to kick the ball out of the protected zone. If a robot fails to clear the ball, the ball will be moved to the opposing team’s attacker zone. 

The game consists of 2 halves. Each half begins with a kickoff where the ball is placed in the center of a goalie zone. Additionally, after each goal, a kickoff is awarded to the team that did not score the goal. 

Protected Zones
In WRO football (a real life LEGO robot football competition), there are no protected zones. Consequently, opposing robots frequently get stuck on each other. WRO football involves lots of starting and stopping to reset robots that are stuck on each other which disrupts the flow of the game.  

[image]

QuirkyCort, 2020-12-11:
Great work! I really like the idea of this soccer game. Seems to me it's easy enough for a beginner to get started on, but complex enough to have a lot of room for different strategies. Having some ways to send messages between robots of the same team would add even more to the strategy (...that's on the TODO list, but I been too busy lately to get around to it).

I'm not familiar with WRO football, but from a quick look around on youtube, the physical robot game doesn't look that interesting. Lots of robots ramming themselves straight into the ball and hoping for the best. I suspect that the protected zone will prevent that and encourage deeper strategy. The mecanum wheels common in WRO football looks like it'll be useful. No way to have that in Gears right now, but it's something to look into.

sgmurray, 2020-12-24:
I now have an arena game mode on my local copy. 

https://youtu.be/HfbHFspoXpE

The arena game mode does include both invisible fences and a 10 second shot clock (see first post). However, since I programmed the example robots to not drive past the border and kick the ball quickly, these features are not explicitly demonstrated in the video.  

Is this something the GEARS community would like to see in the public online version? 

If this makes it to the public version, would anybody be interested in doing beta testing?

QuirkyCort, 2020-12-24:
Looks great. As it is, it already look very interesting and with potential for a lot of strategy.

I don't have a suitable group of students to play-test this, but looking at the video, I would suggest...

- Increase the size of the goal. This will encourage the goalie to adopt better strategy than spinning in place.
- An alternative positioning of the robots, with both robots in the same area. This will encourage more strategy such as pass and shoot and switching between formations (eg. defend-defend, attack-attack, attack-defend). It will also force the coders to consider how to avoid crashing into their team mate.

[image]

sgmurray, 2020-12-24:
I agree that the game should encourage the goalie to do something other than spin in place.  Increasing the width of the goal is one possible solution.  I am thinking of making the width of the goal a user settable option. 

I also agree that the game could do more to encourage coordination among teammates.  However, I think having 2 robots in a defensive zone encourages parking 2 robots in front of the goal. 

How do you feel about a 3v3 game? Same field as before, each team would have 2 attackers  and 1 goalie.

shravan01, 2020-12-25:
Hello,
    This is an amazing arena setup. How is the ball kick done? I also like
the timer setup. This would be a great add on for all competitions.
Warm regards
Shravan

MINTGenie, 2020-12-25:
We saw that by setting the magnetic actuator to negative value the object is repelled. We also noticed that the effect works even when the object is not touching the actuator

QuirkyCort, 2020-12-25:
With 2 robots in the same zone, I would think that the teams will have the following strategic decision to make (...and switch between):

defend-defend:
Two robots in front of the goal. With a wide enough goal, this will be a strong, but not impervious defence. 
- Good: Strong defence.
- Bad: If the opponent manages to score, they have a high chance of picking up the ball again and getting another chance to shoot. If the defender manages to catch the ball, it's a long way to the mid field, giving the opponent more time to get into a defensive position.

attack-attack:
Two robots at the front.
- Good: Pass and shoot to get around the opponent's defence.
- Bad: If the opponent catches the ball, the goal is defenceless, and it may be a long travel to defend the goal.

attack-defend:
One robot at the front, another infront of the goal.
- Middle of the road option, with no particular strength or weakness.

A 3v3 game would also be interesting, but there may be less option for strategy, since it is pretty much fixed into a 2 attacker, 1 goalie formation.

sgmurray, 2021-04-05:
I took Cort's suggestion and changed the game to 2 zones with 2 robots per zone. I also made the example robots a little smarter too. Currently, this is only on my local copy, but I plan to submit a pull request soon.

https://youtu.be/F7duAoTENDo

As always, I am very interested in any feedback the GEARS community has to offer.

QuirkyCort, 2021-04-06:
Amazing work. The robot behaviour looks remarkably human-like with a clear strategy.

Looking forward to the PR. Would love to organize some kind of an online competition with this.

QuirkyCort, 2021-04-09:
Thanks. I've merged the PR. I haven't tested it much, but it looks great.

A few questions and comments...

1) Would you be willing to share your soccer robot design as well? I'll add it as one of the built-in option.

2) The soccerfield image is rather low res, do you have a vector source for it? If not, I can always redraw it and export it at a higher resolution.

3) The green in the soccerfield image is at 70% opacity. Any reasons for that?

4) There may be a better way to separate the two sides using a full wall that lets the ball pass through, but block everything else. This collision mask feature is not well documented (...I'm not sure if it's even available in the 3D engine), and I'll look into it when I get some time.

5) There's a rollingFriction setting which may be a better choice than the current damping and friction setting. Again, I'll look into it when I get some time.

sgmurray, 2021-04-09:
1. I would love to have the robot design included with the built-in options! I haven't submitted the robot yet because I am refactoring some of the code that the design of the robot depends on. My plan was/is to submit ~4 smaller pull requests. The first one is the field. The second would be the code that allows user code to get the position / velocity of the ball and enemy robots (and potentially other user defined objects). The third would be a damping setting to the electromagnet that allows for quicker centering of the ball. The fourth would be the robot design. 

2+3. I am not an artist and have no loyalty to the current soccerfield image and its opacity. I downloaded it from: https://commons.wikimedia.org/wiki/File:Soccer_field_-_empty.svg . I probably should have used the svg. 

4. Sounds good. I was unaware of the collision mask feature. I agree the current dividing wall is less than ideal. 

5. Sounds good. I was also unaware of the rollingFriction setting.

QuirkyCort, 2021-04-09:
1. I used damping on the magnetic items to reduce the oscillation, but it causes other problems. Damping affects all movements of the ball, and if set high enough, it could even affect the movement of the robot that's carrying the ball.

I'm guessing that the problem you're trying to solve is that the ball oscillates for a long time when it hits the magnet. The low friction of the robot body means that any left over kinetic energy in the ball takes a long time to dissipate as it slides back and forth. I've not tested this, but a possible alternative solution may be to place blocks surrounding the electromagnet. The robot body has a very low restitution, so when the ball hits these blocks, it should lose energy and slow down.

2. I've touched up the image a little, and snuck in a gearsbot logo while I'm at it.

sgmurray, 2021-04-09:
You are correct about the problem that I am trying to solve. But I think the way that I am trying to solve it does not affect all movements of the ball. 

The snippet is from robotComponents.js The idea is to turn the 0.1 magic number into a user configurable parameter similar to self.options.maxRange.

[image]

QuirkyCort, 2021-04-16:
I've rewritten much of football world to use World Base. Reduces code duplication and is a little more readable than the older way of doing everything in the world_xxx.js itself.

I've also...
- Rotated the world around.
- Set collision masks and filters to let the ball pass through the invisible center fence.
- Detect when the ball goes out of bounds.

Unfortunately, I can't figure out how rollingFriction works. I've tried setting it, but there don't seem to be any effects.

In the process, I may have inadvertently change some behaviour or introduced some bugs. Let me know if you spot any.

sgmurray, 2021-04-16:
I submitted a PR for some minor fixes. 

I have found that before hitting play if you load a robot from a file (Robot>Load from File), a small ghost image of the ball is left behind. 

[image]

QuirkyCort, 2021-04-16:
That should have been fixed by an earlier commit and I'm not able to replicate it anymore. I forgot to empty objects[] before pushing new objects into it.

Try a reload and see if the problem persists.

sgmurray, 2021-04-16:
It works now. Thanks.

sgmurray, 2021-04-23:
I got the following question on the most recent video. In the interest of consolidating the discussion I am answering it here.

> Wow... this looks cool. How are the blue and green (or red and yellow) passing the ball to each other? which sensor is used for this? could you please share the program for this one.

Robots see each other using the currently  undocumented ObjectTracker which is currently only available in python (sorry Blockly). In the football World, you can also get the position and velocity of the ball

[image]

My vision for the Football World is to be an educational experience. I want students to start from square one and slowly iterate towards more and more interesting solutions. Consequently, I am hesitant to post a complete solution online. That being said, I am willing to share the code privately with educators e.g A Posteriori and MINT Genie to serve as a "teacher's edition".

sgmurray, 2021-04-27:
@QuirkyCort I finally got around to changing all of my user code to work with the new world coordinate system. 

1.  Robots can drive up the invisible fence using their wheels. Setting the friction of the invisible wall to 0 seems to fix this problem.

2. Would you be willing to make robots on the same team closer in color? I know having different colors makes matching print statements in the console to their corresponding robot easier. But, I think having robots on the same team wearing matching colors makes it feel more like a sporting event. 
[image]

shravan01, 2021-04-27:
Thanks this is interesting to know. I would love to prepare a setup and get
the hang of it and prepare some kind of document which students can follow
along - in incremental steps.

Warm regards
MINT Genie Team

sgmurray, 2021-04-28:
@shravan01

I have attached a json file of the robot from the example video this post.
The world is currently part of the online version of gears. Worlds>Select World > Football Arena

@QuirkyCort 
Would you like me to submit a PR for adding the example robot to the samples? 
My sample robot has 2 "casters" where the second caster is a sphere block. I am not sure if the physics of the block are the same as a caster.

[kdamp (1).zip](https://github.com/QuirkyCort/gears/files/6388235/kdamp.1.zip)

QuirkyCort, 2021-04-29:
I've changed the friction to zero for the invisible wall.

I'm planning to add a robot color setting for the arena, with the following options:

**Individual**
Same as what we have now.

**Team**
This one is a little trickier. I'll like it to be...
* Easily differentiable between teams (eg. red vs blue)
* Differentiable between robots within a team (ie. the two red robots should not look identical)
* Color scheme should be extendible to more than 2 teams and more than 2 robots per team.
* It shouldn't look terrible...
I'm thinking of maybe having the same hue in a team and varying the saturation, or having a base hue in a team and offsetting it by a small angle for each team member. Suggestions are welcome here.

**Custom**
Use the color of the robot in the uploaded zip file. This will also let teams embellish their robot with their team logos.

I've added the football robot to the default list. Not need for a PR unless you want to change something.

The default caster doesn't have any special physics, but two caster can be a potential problem. Strictly speaking, a robot with 4 contact points to the ground is over-constrained. Any unevenness (eg. due to numerical inaccuracies) could result in one wheel floating above the ground.

It probably won't be an issue as the joint between the wheel and the body isn't perfectly rigid (...although that appears to be an artefact of the physics engine and not by design). But just in case, I've shifted the added caster very slightly upwards (0.5mm) to ensure that this won't be an issue.

QuirkyCort, 2021-05-06:
Added options for robot colors in the arena.

sgmurray, 2021-05-06:
@QuirkyCort  I love the ability to set colors.  It is a small tweak, but I think it makes the game much more engaging.  

I know this is a nitpick that belongs at the very bottom of the todo list.  Currently, when you hit the checkbox on the bots tab in arena mode the background becomes the bot's 'individual color'. It would be cool if the background matched the color of the robot in team color and custom color mode too.

vsalcaidesfaria, 2023-10-27:
Hi. Is there any progress of this project. 
There is in the future the possibility to use objecttracker in blocks?

sgmurray, 2023-10-30:
> Hi. Is there any progress of this project. There is in the future the possibility to use objecttracker in blocks?

Here is an unofficial version of the GearsBot site with a object tracker block. [https://sgmurray.github.io/gears/public/](https://sgmurray.github.io/gears/public/) . Feel free to offer any feedback that you have.