QuirkyCort/gears #34: Arena Mode Issue

sgmurray, 2020-12-13:
I am seeing a difference in arena mode between the version in the github repo and the version at https://gears.aposteriori.com.sg/arena.html 

I have a robot/program that works in the single player simulator mode.  When I upload the corresponding zipfile in arena mode running on my local machine I get, "ImportError: No module named time on line 4" in the console in the arena.  When I upload a zip file to the aposteriori website arena, the program runs without issue.

Anybody else having this issue?

QuirkyCort, 2020-12-14:
A recently merged PR wasn't properly tested, so I didn't upload it to https://gears.aposteriori.com.sg.

If you need to run a local copy, try rolling back the latest commit e2725a7.

QuirkyCort, 2020-12-19:
The arena bug should be fixed now in both in github as well as on the main site.

closed 2020-12-24 by QuirkyCort