QuirkyCort/gears #35: CRITICAL bug?

DJuego, 2020-12-19:
Hi! I am working with the c5fd654 commit (the last commit) locally. Microsoft Windows 10 x64 and Google Chrome 87.0.4280.88.

Well. File -> New Program -> Confirm?/Ok.

Gears gets locked/blocked! :-D

DJuego

.

QuirkyCort, 2020-12-19:
Thanks. The bug was introduced by a recent PR for Python code separation. My bad for not checking it thoroughly before merging.

I've put in a fix, and it should be fine now. I see another bug (...old Python module remains even after starting a new program), but that's just a minor annoyance (...I'll fix it eventually). Please do report if you find any other issues.

DJuego, 2020-12-19:
I confirm it. Problem solved! Thanks again for your speed and support!

DJuego

closed 2020-12-19 by DJuego