QuirkyCort/gears #36: Dynamically adding blocky blocks for python library defs

humbug99, 2020-12-20:
Feature request: parse python library modules as they are imported and add blockly blocks for python functions.

This was discussed in issue #5 , I still think it's a good idea. (Please refer to that issue for the initial discussion about this feature.)

Following the Makecode annotations seems like a good idea too.  For un-annotated functions, and those with no category, should there be a single default "Python Library" category that they go in, or does each library module get it's own category?

I propose that only top level / non indented defs get processed, in order to avoid all the defs nested inside objects.

I'm willing to try making an initial implementation of this, but I think it may take some time.  I see that the blockly version in use in gears is pretty old, old enough that the docs no longer match the API, so I think I'll need to update the blockly to a recent version.  If there are any issues you can see with doing this, or a particular blockly version you suggest updating to, please let me know.

QuirkyCort, 2020-12-20:
Eh... The version of Blockly in use is just a few months old. That's not that bad. If updating it to the latest, please test very thoroughly as this has the potential to break many things. Separate out the PRs too. It better to have many small PRs, rather than a single big one.

I'll suggest...

- One PR for updating blockly. This one needs to be tested very thoroughly.
- A second PR that generate blocks while ignoring annotations. All blocks goes into a default "Python Library" category. This should be a bare minimum proof of concept. The smaller the code the better.
- Other PRs that adds in additional functionalities (eg. annotations).

This can get pretty complicated, and have very limited use for most users, so it's hard for me to spend time reviewing it unless it's simple.

humbug99, 2020-12-20:
Understood about the blockly update.  Will put this on low priority for now.