QuirkyCort/gears #37: Improvements to Pen

humbug99, 2020-12-20:
I like the changes to the pen, especially being able to have multiple pens!

I found some issues where some things could be improved, some of which I can start working on, and some which need discussion:

1) Changes to pen color affects all traces from the pen, even after a pen up/down pair.  It would be better if either:
- pen color change only affects the current trace and subsequent traces, or
- pen color change takes effect immediately.  This can be achieved by having a color change be equivalent to an up/down sequence.  (I prefer the second option)  [ Looking at the code, I think I see what the problem is, I'll make a fix + PR for this. ]

(The current behavior is not consistent with changes to the pen width, which take effect immediately.)

2) Using the pen in blockly on the default robot fails with an error message in the simulator console.  For the sake of beginners, it would be nice if dropping a pen down block into the blockly code just worked.  To make this happen, the default robot needs a pen, and the default pen blocks need to either use the pen port for the current robot or the pen port for the default robot.  (An alternate way of doing this would be to add an addPen block to blockly, with a port number, that puts a pen on the current robot...)

3) In my opinion, the pen is most useful centered on the wheel axis.  I propose that:
- The robot configurator should have a checkbox that centers the pen on the wheel axis, and we add this flag to the pen config.  This should override the x and z config values, but not the y value.  

4) The wiki should say something about using the pen.

5) Pen sample code needs to be updated to be consistent with the pen interface.

6) Is there some way to have the pen Y (height) coordinate be relative to the world and not relative to the robot?  That is, suppose I want to draw the pen line at a particular height.  To do so, I have to compute the robot body height based on the robot body, the wheel diam, and the wheel offset, and then the pen height will change if I modify any of these.  It would be much simpler if there was a way to just set the absolute pen height, and then have the pen component compute the correct y placement to get this height.  (Maybe this is another flag in the options + checkbox?)

I will work on 1, 4, and 5 above.  I'd like to get some discussion on 2, 3, & 6 before I do anything about these.

QuirkyCort, 2020-12-20:
2. I'm not in favor of blocks/command that magically add parts to the robot. It's not consistent with the rest of the components and it's hard to draw a line between which components needs it and which doesn't. Adding it to the default robots make sense; I think all of the robots will benefit from having a pen.

Getting the port correct is an issue for all of the components. Right now the user must set them every time and it can be a pain. I'll add an "Auto" option to all the port settings and make that automatically choose the correct port (...provided there's only one component of that type).

3. Your control loop, whether it's PID or something else, is actually controlling the position of the color sensor. If you're interesting in investigating the stability and response of your control algorithm, the pen should be right on the color sensor (...if using one) or between the two color sensors (...if using two).

It's not hard to manually position the pen on the wheel axis, but if there's a demand for automatic positioning, then I would think that a better approach is to set the correct x/y value. Possibly a drop-down to let the user choose between a few pre-sets. No need for any special flags or changes to the pen code.

6. Same as before. Probably a button or dropdown to set the xyz accordingly rather than a special flag.

humbug99, 2020-12-20:
Having an Auto option for blockly ports is a great idea; I think this in combination with having a pen on the default robot(s) will make it easy for blockly users to get the pen working with little effort.  I'll add a pen to the default robot in preparation for this.

For 3 & 6, having a button/dropdown to move the pen to a particular preset solves the issue, I just wonder if there is some UI decoration that can help users to understand that they will need to use the button/dropdown again if they change the dimensions of the robot.  Maybe if the pen is already in the location connected with a moveto button, that button is disabled? 

While we're talking about the pen, I'm curious as to why you decided to give the pen a physicsImpostor?  It seems to me that if you are thinking about use pen rotation to be able to draw outside the x-z plane (which is a nice idea), then some use cases might require some funky pen positions, in which having the pen be part of the physics calculations might result in the pen dragging on the floor or on objects in the world.  (No particular use case in mind here, just wondering...)

humbug99, 2020-12-24:
I started to make some documentation for the Pen to put in the wiki.  I see that the wiki isn't part of the repo.  In what form would you like suggestions for wiki content?

I'm going to try implementing buttons for 3 & 6...

QuirkyCort, 2020-12-25:
You can email it to me. cort@aposteriori.com.sg

closed 2021-03-23 by QuirkyCort