QuirkyCort/gears #39: Question: How to add timers

MINTGenie, 2020-12-25:
Is it possible to add a timer or a stop watch or some counters to custom worlds. We felt the need for this during our competition.  Your suggestion would be helpful here.

QuirkyCort, 2020-12-25:
Unfortunately, there are currently no ways to add a timer to the custom world. I'm gradually adding more capabilities to the image world and timer is likely to be added. But in the meantime, if you need more customization than what's currently possible, the best way is to create a new world in javascript.

QuirkyCort, 2021-03-16:
There has been a big revamp of the custom world system and there is now an option to add a timer. Best way to add the timer is via the new World Builder.

The timer can be set to run upwards from zero or in a count down mode. Duration is configurable, and you can force the robots to stop automatically when the time runs out.

closed 2021-03-22 by QuirkyCort