QuirkyCort/gears #40: Little suggestion: its own programming block for Laser Range Sensor?

DJuego, 2020-12-28:
I admit that this is a small, low-priority annoyance.

I wish the laser had its own sensor block. I find it more coherent and elegant, in parity with the other sensors. Also, when reading the program there would be no risk of confusing a laser reading with an ultrasound reading.

I understand that, in essence, it behaves exactly like the ultrasonic sensor  (as explained in the wiki)  but it is still a different sensor that even has its own graphic model (as it should). After all, they rely on very different technologies (sound and electromagnetism) :-D

DJuego

QuirkyCort, 2021-01-04:
Added a block for it.

I'm generally not keen to add blocks which only has a cosmetic difference, but in this case, it helps with the auto-port selection.

closed 2021-01-04 by QuirkyCort

DJuego, 2021-01-04:
Thank you for this. @QuirkyCort !  :heart:

In My Humble Opinion, even if the code under the block was exactly the same,  it would not be simply a cosmetic issue. It has didactic connotations and reflects reality more accurately (they are different sensors).

Thanks again!

DJuego

DJuego, 2021-04-01:
Hi again, @QuirkyCort ! Today I noticed that in python there is no laser sensor. I propose to add it for the same reasons as the laser block.  Of course there is no default laser sensor for Lego. It would be a "virtual" sensor, I think.

DJuego