QuirkyCort/gears #41: The color sensor cannot "see" the paint splatter...

DJuego, 2020-12-29:
It seems that the color sensor cannot see the paint. Would it be feasible to solve this? Maybe it could even be an optional feature (in PaintballLauncher options for example) if you don't want it to affect algorithms that follow lines on the floor...

DJuego

humbug99, 2021-01-03:
It would be nice if the robot could optionally see the pen traces also - the robot could leave a pen trace behind itself, then follow the pen trace back to the start point...

QuirkyCort, 2021-01-05:
This can possibly be implemented as an option in the paintball launcher and pen.

The color sensor is very computationally expensive to run; each color sensor is essentially a camera rendering a new image on every read. Some savings are achieved by pre-caching all the shaders, but since the paintball splashes and pen traces are added dynamically, they are not included in the cache.

The paintball splashes shouldn't add much to the computational load, but the pen traces is currently rebuilt from scratch every time it is extended, and that will invalidate the cached shader and requiring it to be recompiled. Could be problematic on slow computers.