QuirkyCort/gears #49: Unexpected behavior: An object is always "visible" to the ultrasound/laser even if it has the physics turned off and/or is not visible.

DJuego, 2021-01-05:
It is possible to make an object invisible by changing the transparency color. 

It is possible to disable the physics of an object with: "physicsOptions" : false
HOWEVER
An invisible object with physics turned off is still visible to the ultrasound/laser 😲 ! This is a bit unexpected.

Anyway, it would be interesting to have an additional option (in JSON) to make an object invisible to ultrasound/laser. This way, objects with materials and/or surfaces that are problematic for these sensors could be simulated.

DJuego

QuirkyCort, 2021-01-06:
An oversight on my part.

I've added in two options: laserDetection and ultrasonicDetection. Each can be set to...

"invisible" : The rays passes through the object as if it is not there. Object cannot be detected, but the sensor can detect object behind it.

"absorb" : The rays are absorbed by the object with no reflection. Sensor cannot detect objects behind it.

"normal" : The rays are reflected back normally. Object can be detected.

If not set, these two new options will default to "normal" for an object with physics and "invisible" for a physicsless object.

DJuego, 2021-01-06:
This is so much more than I expected. You are amazing! I'm really sorry to bother you with these details, and I hope to compensate you in the future in some way. 

The flexibility is wonderful.  I just hope my suggestions aren't making Gears too complicated for newcomers. Or for the future scenario editor. 

If you think so, tomorrow I will test these new capabilities and if everything goes well I will close the thread. 🍶 

DJuego

DJuego, 2021-01-09:
Well. I have tested the laser and ultrasound in all three modes and everything seems to be working as expected. I would say that the first quality control has been passed. :-) Thanks again for this! It brings a level of detail to the environments.

DJuego

closed 2021-01-09 by DJuego