QuirkyCort/gears #50: The Pen is an actuator. Why is a sensor port (inX) assigned?

DJuego, 2021-01-09:
Thank you very much for the Pen! It is a useful educational tool. 

However, there is something I don't understand: why is it assigned a sensor port (in1, in2, in3m...) and not an actuator port (outA, outB, outC,...)?

DJuego

QuirkyCort, 2021-01-10:
Actually, input ports are not purely input only and the output ports are not purely output. For a Mindstorm robot, the inputs ports are bidirectional digital data ports (I2C and/or UART, plus some others), and the output ports have both motor driver outputs and an input for the rotary encoder readings. In the actual EV3, the input ports are sometimes used to send out signals (eg. Ultrasonic can be used to send signal to another robot) and it's common to read the motor positions from the output ports.

**So to me, I see the "input" ports as general digital data ports, and the "output" ports as motor control ports. Anything that can be modelled as a single motor (eg. arm, swivel, paintball launcher) goes to the output ports, and everything else goes to input.**

\* While it is technically possible to implement the pen (...with up/down, colors, and width settings) with either the input or output ports, it would be extremely difficult to do so using the output ports as the only means of outputting data is by motor direction and power. With the input port, it is actually practicable with some custom electronics. Won't be able to set any arbitrary colors or line width, but switching between a few different pens isn't hard.

DJuego, 2021-01-10:
Thank you very much for your extended explanation! Very interesting!

Yes, I knew that from the input ports it was possible to make outputs. Although I have always felt it as an exotic use. And of course it is well known that the tachometer reading is obtained from the output ports. However I had always taken quite seriously the terms input and output because .well... those are their names ("in1", "outA"). :grin: From the point of view of didactic purpose and simplicity it is great. But it is also true that sooner or later they will have to realize that the world is more complicated.:grinning:

 Based on this new information it would be more appropriate to call them "motor ports" (It is true that sometimes they are known this way) and "a more general purpose ports". 

 From what I see your philosophy with Gears is to approach the real electronics/limitations of the EV3 units instead of considering an "ideal" robotic unit. And so imagine where it would be easier to hook a "hypothetical Pen". Ok!

DJuego

closed 2021-01-10 by DJuego