QuirkyCort/gears #52: Bug: "Empty" endless loop produces error.

DJuego, 2021-01-29:
Hello again!

I'm pretty sure I "reported" this problem (or very similar) in an issue. And it was successfully resolved... but it's back! :-O And now I can't find that old issue. :-O

This code takes a few seconds to start executing and produces an error in the console (current master) 

[image]

I get  in console: _"TimeLimitError: Program exceeded run time limit. on line 35"_

DJuego

sgmurray, 2021-01-31:
Try adding a sleep inside the loop.

[image]

DJuego, 2021-02-01:
Thank you for your reply!

Indeed, sgmurray .  I knew that. The thing is that I have memories that @QuirkyCort certified that this was not the expected behavior, and that the loop should work even without the sleep. He even solved it. But now I am unable to find the issue where this was discussed. Did I dream it? :sweat_smile:

DJuego

QuirkyCort, 2021-03-11:
Sorry for the late reply. Been very busy lately.

A sleep is needed because of the way the Python interpreter runs in the browser. Without it, the Python interpreter will loop continuously without releasing control to the browser, and the browser tab will lock up. There's some code to detect this situation and kill the Python program after 5 seconds, generating the TimeLimitError that you've seen.

You don't need a sleep in the loop if it's reading a sensor. This is because all the sensor reads have a small sleep hidden in them. So if you do a "repeat while color = white", it'll work fine.

closed 2021-03-22 by QuirkyCort