QuirkyCort/gears #54: How to set the Robot position in the z axis

Bugsia, 2021-03-08:
Hello,

in our level we want our Robot to start on a elevated platform but in the code there is only the option to move him in the x andy y  axis. Is it somehow possible to keep it starting on the top of the platform?

QuirkyCort, 2021-03-11:
I've added a "z" option to "startPosXY". You can now set the starting position with "x, y, z".

DJuego, 2021-03-11:
Happy "to see" you again! @QuirkyCort!

Related to this issue it would be nice to add the z in "Display current position" and "Save current position to settings".

And, by completeness, also the angles with respect to the three axes. :-D

DJuego

Bugsia, 2021-03-11:
Hello @QuirkyCort,
I just tried out to spawn my robot above the surface, but always, when I try it, it doesn´t work. The world loads nomally, but the robot is still on the ground.
And when its spawned inside a object all of the object dissapear

QuirkyCort, 2021-03-12:
Here's an working configuration that starts the robot 50cm above ground.

```
{
  "worldName": "image",
  "options": {
    "imageURL": "https://i.imgur.com/V5rrel0.png",
    "length": 830,
    "width": 557,
    "imageScale": "2",
    "startPosXY": "-354,60,50",
    "startRot": "180"
  }
}
```
Can you compare the "startPosX" line and see if it's different from yours?

Bugsia, 2021-03-12:
Thank you! Now it works. I just copied your line into my code and it worked. Dont know why it didnt worked before

closed 2021-03-12 by Bugsia