QuirkyCort/gears #55: The robot does not stop after passing the track

CRcharo, 2021-03-11:
We programmed with students following the line last week on the simulator. We used a GPS sensor to measure the distance traveled on the y-axis. Everything worked normally last week. Today, when the specified distance is traveled, the robot does not stop. We tested various browsers and computers, tracks and robot settings on the simulator. Nothing helps. Where is the mistake? Will you help us please? 

[image]
Last week
[image]

sgmurray, 2021-03-11:
@CRcharo For a quick "fix" to your problem, try using altitude instead of y. 
@QuirkyCort Previous versions of virtual.py swapped the sensor's y and z to align with the world frame. The current version does not.

CRcharo, 2021-03-11:
Thanks for quick idea. It seems that it works. In sensors window in simulator it is still Y. That's a little confusing. Thank you again very much.

QuirkyCort, 2021-03-12:
Sorry about that. Made a mistake in an earlier commit. I've fixed it so that y will work correctly now.

CRcharo, 2021-03-12:
@QuirkyCort Thanks a lot. We use your simulator at school to teach robotics, because now our schools are closed because of Covid-19. Really great job!!! Thank you very much again.

QuirkyCort, 2021-03-13:
Glad it is useful for you!

Closing this issue.

closed 2021-03-13 by QuirkyCort