QuirkyCort/gears #56: Firefox but not Chromium

SebCanet, 2021-03-22:
Hello,
I tried to launch Gears directly in Firefox, and everything works fine, without any webserver. But if I try to launch in Chromium there are lot of errors, maybe due to CORS request.
And of course I have the same problem when I launch GEARS through Electron ([https://github.com/technologiescollege/GearsDesktop](https://github.com/technologiescollege/GearsDesktop)).
Firefox 86.0.1 x64 without any tweak.
Chromium 89.0.4389.90 x64

Any idea?

QuirkyCort, 2021-03-22:
I did not test using Chromium, but Chrome is my primary development environment for Gears. I'm pretty sure I'm not using any of the proprietary addition to Chrome, so I would expect them to work the same. Never tried through Electron.

It may be the GPU Blacklist. https://www.khronos.org/webgl/wiki/BlacklistsAndWhitelists

For various reasons, browsers may disable WebGL support (...typically because the driver is old and not updated). Firefox and Chrome each uses their own blacklist, so one may be affected while the other is not. From my limited experience, it seems that Chrome is a little more strict on this than Firefox.

Try disabling the GPU blacklist...

Go to chrome://flags/
Search for "#ignore-gpu-blocklist" <= This is writen as "block" not "black".
Enable the setting and restart your browser

If that doesn't work, please share the error messages (...open up the debug console with  F12), and I'll see if I can figure it out.

QuirkyCort, 2021-03-22:
Just to clarify... Chrome **on Linux** is my primary development environment for Gears. Never seen a problem that occurs only on Windows and not Linux, so it shouldn't matter, but there's always a first time...

germanicianus, 2021-04-13:
@SebCanet, for me Chromium works. I installed it from the official Ubuntu repository. _About Chromium_:
`Version 89.0.4389.90 (Official Build) Built on Ubuntu , running on Ubuntu 18.04 (64-bit)`

closed 2021-08-12 by QuirkyCort