QuirkyCort/gears #59: Bug with laser sensor! It does not work!

DJuego, 2021-04-01:
First; This is a bug detected in the most recent [commit](https://github.com/QuirkyCort/gears/commit/449e62f6706b3b6683e0c3a6ce17d0e45306c741). 

I attach a zip package that includes a robot that only has the laser sensor. Nothing else. The program is empty.

When you try to run it, you get the error:

`TypeError: No ultrasonic sensor connected to in1 on line 197`

Interestingly enough, it happens to be related to my last issue (#58). :kissing_smiling_eyes: 

Cheers!

DJuego

[laser_bug.zip](https://github.com/QuirkyCort/gears/files/6244289/laser_bug.zip)

QuirkyCort, 2021-04-02:
Sorry. This is due to a recent commit that adds a check for the sensor type.

Previously, it was possible to initialize a color sensor on a particular port, even if it was a different type of sensor connected on that port. Won't work, and it gives confusing errors. The newly added check will throw a proper error on initialization, but I missed out adding the check for the laser sensor.

It's fixed now.

DJuego, 2021-04-02:
Yes. I can confirm that this problem seems to be fixed with the latest commit. :satisfied: I close the issue.

DJuego

closed 2021-04-02 by DJuego