QuirkyCort/gears #60: Customizing the ultrasonic sensor..

DJuego, 2021-04-02:
I ignore if the actual ultrasound parameters try to match the actual parameters of the EV3 ultrasound sensor.

```
"type": "UltrasonicSensor",
"options": {
  "rayLength": 255,
  "rayIncidentLimit": 0.698132
}
```
Good. The length of the ray (255), yes.  👍 

I have the feeling that the emission angle now is too narrow for what we see on the real robot. Maybe others with more experience can correct me. 

In any case, It could be interesting to customize the number of rays, and above all, the angle of the "fan". 

DJuego

P.S:
When the wiki says that the rays are emitted in a fan, I understand it to mean that they are emitted in a single plane. It is not a cone.  Right?

QuirkyCort, 2021-04-02:
The number and angles of the rays are in fact customizable; it's just not documented.

```
"options": {
  "rayRotations": [
    [-0.035, -0.305], [-0.035, -0.183], [-0.035, -0.061], [-0.035, 0.061], [-0.035, 0.183], [-0.035, 0.305],
    [0, -0.367], [0, -0.244], [0, -0.122], [-0.035,0], [0, 0], [0.035,0], [0, 0.122], [0, 0.244], [0, 0.367],
    [0.035, -0.305], [0.035, -0.183], [0.035, -0.061], [0.035, 0.061], [0.035, 0.183], [0.035, 0.305]
  ],
  "rayLength": 255,
  "rayIncidentLimit": 0.698132
}
```

This are the full settings. The rays are emitted in roughly a flattened cone shape, like this...

[image]

Also, there are now 21 rays, not 19. I added two more (...slightly above and below the center ray), as it wasn't detecting objects that are right in front but slightly below the ultrasonic (...a common occurance).

The emission angle of the real ultrasonic isn't easily quantifiable, as it depends on the object's material, size, shape, and orientation. Horizontally, I think gears is pretty close, but in the vertical direction, gears is rather narrow.

DJuego, 2021-04-02:
Good. I am pleased with your response. :relieved:  It is quite likely that I will play with those parameters in the future.

Thank you once again, @QuirkyCort 

You can close the issue if you wish. :wink:

DJuego

closed 2021-04-02 by QuirkyCort