QuirkyCort/gears #61: Identify a object that is magnetic

Bugsia, 2021-04-03:
Hello,

I am doing a challenge right now and one of its tragets is to get a magnetic object and transport it back to the strating point.
The transporting is no problem but I dont know a great way to detect the object.
I tried to find it by getting the difference between the old measured distance from a ultrasonic sensor and the current one, to detect if there is an object infront of you but its very unreliable and can be triggered by many other things or doesnt triggers at all.

Is there a better way?

QuirkyCort, 2021-04-05:
I'm guessing your magnetic object is small, so that won't be detected reliably using an ultrasonic. Making it larger will make it more easily detectable, but you might have difficulties picking it up.

A few solutions you can consider...

1) Make the magnetic object larger, and instead of picking it up with the magnet below the robot, configure a robot with a magnet in front of it. The robot can then drag the object around without lifting it up. Depending on the size, you may need to tune the magnet parameters to prevent the object from coming off.

2) Instead of using a magnet, use some kind of a mechanism to capture and move the block. This will let you use a bigger object that can be easily detected by ultrasonic. The recently added cage bot can be used for this, or you can create your own robot.

3) Drive over the magnetic object and detect it using the color sensor.

Bugsia, 2021-04-05:
Thank you! I will look what option is the best.

Bugsia, 2021-04-06:
Hello. It worked really well with the colour sensor and I can detect it now but the Arm ,on which the magnet is placed, moves extremely fast and the magnet cant keep the sphere connected.
In the Robot Configurator there is no option for speed.

So is there a way to slow down the arm?

QuirkyCort, 2021-04-07:
The arm moves at the same speed as an EV3 large motor without gears. If you want it to slow down, just set a lower speed (eg. 5%).

closed 2021-04-07 by Bugsia