QuirkyCort/gears #62: Suggested feature: enhanced development environment for Python?

DJuego, 2021-04-06:
So far I have been using Blocky for education, but I am considering using Python directly as well. For example, workshops aimed at target students in pre-university courses. I have been testing it and it has been, in general, quite satisfactory. :kissing_closed_eyes:

 What I miss the most are autocomplete functions. For example, context menus that open showing the methods of the class when you type a dot after the name of an object.  

I do not know if such a thing is possible o there are open resources available to achieve this. I know it is a complex thing. Certainly that feature would be a **drastic** improvement on the current Gears. 

DJuego

QuirkyCort, 2021-04-07:
The python editor is using Ace and it supports only a rudimentary autocompletion. If anything, it seems almost counter-productive as it pulls in everything as a symbol (...even including comments). The only customization I've done in this area is to add in the method names of the sensor / motors objects, but it doesn't recognize the object type and will offer all the method name every time.

What I see as practical in this area includes...

* Proper autocomplete for methods of Gears objects (ie. motors and sensors).

* Proper autocomplete for common modules (ie. math and time)
 
* A separate help panel which will show contextual help based on where the cursor is (eg. it will show the api docs for motors if the cursor is on a motor object). This will make use of the currently wasted space on the right.

If implemented, there would likely be a number of gaps (eg. object types can't be determined correctly in some situations, no autocomplete support for user created classes).

An alternative would be to switch editor to Monaco (...editor used in VScode) for its full intellisense support. May be a lot of work, and I'm not sure if it's worth the effort; none of my students are using Python with Gears and there may not be that many people using it.

DJuego, 2021-04-07:
Once again thank you for such an insightful response, @QuirkyCort. 

For several years I have done workshops for students in pre-university courses using Lego MindStorms and "EV3dev Python". The development environment I used was WinPython (Spyder is a fantastic IDE for Python). I think it's a great way to introduce students to real programming in a fun way. 

So, now that there is a **World Builder** (amazing!), the only two major milestones I can imagine for Gears are:

1) Pybricks compatibility (and thus getting compatibility with the new LEGO Education SPIKE Prime set and the LEGO MINDSTORMS Robot Inventor sets).

2) An advanced Python development environment. I did not know  [Monaco](https://github.com/microsoft/monaco-editor). It looks very promising. You get the feeling that, in the long run, their integration into Gears would pay off.

I tested Monaco very briefly and did not get the autocompletion of methods of an object (by pressing the dot). maybe it's my mistake?

[image]

DJuego

QuirkyCort, 2021-04-07:
Eh... I tested it out on VSCode and it works fine, but on the online sample, it's not picking up the Python methods. Probably need to configure something to make it work. I'll dig through the Monaco docs if I get around to replacing Ace.

As far as Python API goes, I much prefer ev3dev-lang-python, but I understand that many prefers pybricks for its faster startup time. I have not used the spike prime / robot inventor, but from what I've seen from the docs, it appears to be using a different API from pybricks. Pybricks appears to be in beta for these.

So now there are 3 API (...excluding some of the minor ones), ev3dev-lang-python, pybricks, and Lego Spike Prime Python. It's not terribly hard to support all three of them, but pybricks is particularly problematic, as...

* It lacks move steering and move tank
* I don't exactly know what drive() does

If I'm going to be implementing runnable pybricks support, I'm gonna need some help on clarifying the details for the API behaviour. Either that, or a lot of time to test things out on the physical robot.