QuirkyCort/gears #65: Add documentation on main page regarding security of http.server and binding to localhost only

germanicianus, 2021-04-10:
The main page of this repository states:
> If you do not have a webserver, but have Python3 installed on your computer, you can try changing to the gears directory and running...
>
>python -m http.server 1337
>
>This should get the site running on http://localhost:1337 (...try http://127.0.0.1/1337 if that doesn't work). Do not close the window with the Python command running.

It should be fixed regarding these points:
- If a user wants to have only local users, he should restrict the access using the following command.
`python3 -m http.server 1337 --bind 127.0.0.1`
See https://docs.python.org/3/library/http.server.html
- The statement
`http://127.0.0.1/1337`
must be changed to
`http://127.0.0.1:1337`

QuirkyCort, 2021-04-11:
It's not really a security issue. Gears only contains static files, so even if the http server is open to the public, it'll only expose files that are already publicly available. Nevertheless, I'll add in the bit about binding to the loopback interface.

You're right about the /1337. Typo on my part.

As git is not widely installed outside of programming circles. I'll put in instructions on downloading the zip instead.

germanicianus, 2021-04-11:
Fixed -> closing issue :) .

closed 2021-04-11 by germanicianus