QuirkyCort/gears #68: Layout problems on small screen, e.g. world selection dialog has to large font compared to other UI parts

germanicianus, 2021-04-10:
See screenshot below. On small screens like phones this is a huge problem which OpenRoberta doesn't have.
[image]

QuirkyCort, 2021-04-11:
I've put in a fix for that. The entire center section is scrollable now.

germanicianus, 2021-04-11:
The font and the buttons are still really large compared to the main window, but the scrolling is fixed indeed. I didn't try it yet on real hardware.
[image]

QuirkyCort, 2021-04-13:
Fonts are often not scaled to the screen dimensions, else it may be too small to read.

As long as there's no major usability concerns, I'll rather have larger fonts than smaller. Larger fonts helps with...

- Educators presenting to a class...
- Kids with small screen...
- Users with reading difficulties...

germanicianus, 2021-04-13:
Agreed. But I think there should be some zoom buttons in the UI like the Simulator already has for the 3D representation. The reason is, that there are browsers on mobile devices which don't have controls for that, e.g. Amazon Silk on the Fire TV 4K - and the UI just looks HUGE on a 65 inch TV. Additionally, there should be a fullscreen toggle button for the same reason and also to save screen estate - especially on an iPhone in landscape mode because the tabs and URL bar occupy a lot of space and they cannot be hidden because there is no room to scroll down in the UI (which would hide it).

closed 2021-04-13 by germanicianus