QuirkyCort/gears #71: Show fps

germanicianus, 2021-04-11:
It's helpful for users to know the fps of the simulation, especially on older devices and with different browsers. I _gears_ works on an _Fire TV Stick 4K Ultra HD_ in the default browser _Silk_, but feels a bit sluggish and the Bluetooth mouse is a bit laggy. When I use _Firefox for Fire TV_, the world feels quite smooth and the mouse is also more responsive, but the browser is soon to be discontinued. Therefore I installed and tried [Fennec from F-Droid](https://f-droid.org/de/packages/org.mozilla.fennec_fdroid/) which is based on the latest Firefox for Android. It is really responsive regarding the Bluetooth mouse (and requires one), but the world feels really slow.

I attached a patch ([gears_show_fps.zip](https://github.com/QuirkyCort/gears/files/6292145/gears_show_fps.zip)) with a hacky solution to show the fps which can be used as a starting point.
[image]

QuirkyCort, 2021-04-13:
Added an option to display FPS. You can find it under Help.

germanicianus, 2021-04-13:
Great! I just tested it. :)

closed 2021-04-13 by germanicianus