QuirkyCort/gears #73: Robot Configurator not working on Android - tested with latest Amazon FireTV 4K and phone

germanicianus, 2021-04-13:
The controls are shown, but the rest of the screen stays white.

On an iPhone 7 Plus with latest iOS it works - despite some layout problems.

QuirkyCort, 2021-04-14:
Can't tell what's the problem from that, but it may be because of the virtual joystick; it tries to set it up in the configurator, and throws an error because it is not present. It doesn't seem to cause a problem when tested on my browser, but different browsers may handle these errors a little differently.

I've put in a fix. See if it works now.

germanicianus, 2021-04-14:
@QuirkyCort, it works now. It would be very helpful when you reference the issue(s) in the correspondent commits for tracking. Then the commits would be listed within the issues automatically. For examples see https://github.com/plone/buildout.coredev/issues/67.

For the record - this issue is fixed by https://github.com/QuirkyCort/gears/commit/abbaa16191134e99e078f07f96cf6adbb4e9f288.

closed 2021-04-14 by germanicianus