QuirkyCort/gears #74: iOS: Links to Robot Configurator and World Builder not working

germanicianus, 2021-04-13:
After clicking on the respective menu entries and confirming with a button click in the opened dialog on _Go to Configurator_/_Go to World Builder_, nothing happens.

When manually opening the correspondent links in Safari, both open and the UI works.

Tested on an iPhone 7 Plus with the latest iOS.

QuirkyCort, 2021-04-14:
May be an issue caused by popup blocking. The link is opened after the dialog window closes, and not immediately when clicked. Some browsers may not like that and may block that as a popup.

I've put in a fix. See if it's ok now.

germanicianus, 2021-04-14:
Works now. Fixed by https://github.com/QuirkyCort/gears/commit/a222ba77ee19f7e9f41d79de3fa62379b8dfe420.

closed 2021-04-14 by germanicianus