QuirkyCort/gears #75: Pause/stop Babylon rendering when Simulator invisible to save battery

germanicianus, 2021-04-13:
This is really important for mobile devices like tablets used in schools for education. ~ 40 % of my CPU are used when just the block view is visible and I do nothing. When I used _gears_ on my phone, ~ 10 % of my battery were drain in a very short time.
[image]

QuirkyCort, 2021-04-14:
Modified it so that the 3D rendering will only run if the simulator tab is opened, or if the user's program is running.

germanicianus, 2021-04-14:
Works -> Close issue. Implemented with https://github.com/QuirkyCort/gears/commit/cf1ae84c1338b441f9b724980bf68834ce66c44a.

closed 2021-04-14 by germanicianus