QuirkyCort/gears #76: Add UI control to reduce rendering resolution for low end or high density devices

germanicianus, 2021-04-13:
Reducing the rendering resolution makes it possible to either save battery or to increase FPS - depending on the needs.
Examples with default world - approximate FPS:
- Amazon Fire TV on UHD TV: 30 FPS
- Android phone with HD screen: 30 FPS
- iPhone 7 Plus: 60 FPS
- Laptop with HD and iGPU: 50 FPS fullscreen, 60 FPS on 1/4 browser window size

Some info (if needed):
https://www.html5gamedevs.com/topic/22907-how-to-control-enginecanvas-screen-resolution/

QuirkyCort, 2021-04-14:
I've put in some optimizer code, you can enable it from the console with "babylon.optimizer.start()".

From my tests on a slow laptop, the optimizer didn't make much of a difference in the FPS (...from around 30 to around 40), despite a very aggressive (...and ugly) reduction in resolution. More importantly, the robot behaviour didn't improve and seems even worse, but I haven't done enough testing for this to be conclusive.

The biggest issue with slow systems is that the robot tends to miss detecting lines using the color sensor. In game worlds like WRO or Fire Rescue, this means that it'll miss detecting a junction. This is a far more serious problem than FPS and is the primary concern for any performance related changes.

germanicianus, 2021-04-14:
@QuirkyCort, do the visual rendering and robot simulation run independently? I didn't have an in-depth look into the code. I ask because you wrote:
>The biggest issue with slow systems is that the robot tends to miss detecting lines using the color sensor. [...] This is a far more serious problem than FPS and is the primary concern for any performance related changes.

QuirkyCort, 2021-04-15:
They are not independent. The robot simulation requires the rendering to take place, as it relies on the rendering to determine what the color sensor sees. There are many other places where the rendering and simulation are linked, but those can be separated with (...a lot) of work.

closed 2025-05-29 by QuirkyCort