QuirkyCort/gears #77: Visuals crash when a function is called, Code continues

Bugsia, 2021-04-14:
Hello,

today when I started my Code, at the beginning everything worked fine. But as soon as I try to Pickup a magnetic Ball/call a Function that does that I can´t control my camera anymore. It is just frozen. I can see the outputs of the code on zhe bottom, so the code seems to continue running.

It just started today. Can somebody help me?

Thanks
[Code.zip](https://github.com/QuirkyCort/gears/files/6309118/Code.zip)

QuirkyCort, 2021-04-14:
Sorry. Some enhancements to the magnet added a bug. It only occurs when the magnet is not directly attached to the main body, so I didn't notice it earlier.

I've already fixed it, so it should be fine now.

Bugsia, 2021-04-14:
It doesnt really work. The arm now moves to the ball bu then it does the smae thing. And I am also still using the image map and not the custom one, because the start position of the Robot doesnt seem to work the same way, as it spawns at 0,0,0 and when I change them all of the objects are gone.

QuirkyCort, 2021-04-14:
You'll need to provide a bit more details than that.

1) Make sure you click the browser's reload button to ensure that you are using the latest copy and not one from your browser cache.

2) Press F12 to open up the browser's developer console.

3) Run until you get the error.

4) Let me know what error messages appear in the developer console.

Can you also provide more details about your problem with custom world? Did you create it using the world builder? Or did you write it manually? Some of the options have changed name, so if you're writing it manually, perhaps the problem occurred because you're using the old option name.

Bugsia, 2021-04-14:
It works again.
Thank You!

closed 2021-04-14 by Bugsia