QuirkyCort/gears #81: Unable to upload zip files to Arena

sgmurray, 2021-05-05:
Uploading a zip file and clicking the checkbox in Arena mode bots tab gives an error. 

[image]

QuirkyCort, 2021-05-05:
It's a race condition. Sometimes the render function will run before the robot is fully loaded.

I've put in some additional checks to prevent this. Should be ok now.

sgmurray, 2021-05-05:
I am still able to trigger the race condition. 

self.leftWheel != null seems to work better than typeof self.leftWheel != 'undefined' in Robot.jsj

QuirkyCort, 2021-05-06:
You're right. self.leftWheel is always defined, so I should be checking for null instead. I've fixed it.

Strangely enough, I couldn't trigger the problem after adding the type check. Probably just a coincidence...

closed 2021-05-08 by sgmurray