QuirkyCort/gears #83: "invalid path" problem in Git Repository

DJuego, 2021-05-12:
Hi!! It seems that since two days ago there is a problem in the repository that prevents to synchronize with the master...

```
error: invalid path 'public/models/Devil's Garage: Chess/License.txt'
error: invalid path 'public/models/Devil's Garage: Chess/bishop_01.glb'
error: invalid path 'public/models/Devil's Garage: Chess/king_01.glb'
error: invalid path 'public/models/Devil's Garage: Chess/knight_01.glb'
error: invalid path 'public/models/Devil's Garage: Chess/pawn_01.glb'
error: invalid path 'public/models/Devil's Garage: Chess/queen_01.glb'
error: invalid path 'public/models/Devil's Garage: Chess/rook_01.glb'
error: invalid path 'public/models/KayKit: Dungeon/License.txt'
error: invalid path 'public/models/KayKit: Dungeon/arrow.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/artifact.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/axeDouble_common.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/axeDouble_rare.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/axeDouble_uncommon.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/axe_common.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/axe_rare.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/axe_uncommon.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/banner.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/barrel.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/barrelDark.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/bench.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/bookA.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/bookB.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/bookC.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/bookD.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/bookE.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/bookF.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/bookOpenA.gltf.glb'
error: invalid path 'public/models/KayKit: Dungeon/bookOpenB.gltf.glb'
[...]
```

DJuego

QuirkyCort, 2021-05-13:
I think windows don't like the colon in the directory names (...no issues on Linux). I've renamed them. Let me know if that fixes it.

DJuego, 2021-05-13:
Good catch! 

Indeed. I work in Windows 10. I should have realized that the problem lay in the use of special characters in folder names. Now it syncs great! Thanks for solving it! I close the thread. :-)

DJuego

closed 2021-05-13 by DJuego