QuirkyCort/gears #84: Congratulations,

dancito20, 2021-05-13:
Congratulations,
I am a computer teacher from Honduras in a small town in Central America, I am very happy with this program I have implemented it with students from 2nd to eleventh grade and it has given me very good results, on behalf of all the boys I thank you for so important collaboration with education mainly to countries with few resources to obtain a simulator license.
A recommendation that I can give you is that you add the functionality that allows you to move the sensors to the position where it is needed or that we can create our own robot designs I do not know how difficult it is but if it is at your disposal to be able to import the models made in LEGO DIGITAL DESIGNER OR STUDIO would be a very good update, there is an error that generates the program and that is when we want to remove the sensor block when it is joined with the moperator block, the same there generates a problem and the program is damaged, also the distance sensor does not It detects the blocks well, only the edge of the table, I will leave you some videos of some tutorials that I have made to my students, thank you very much for this contribution.
att. mrmartinez@columbusschoolhn.com

dancito20, 2021-05-13:
I would like to work improving the software I do not have much experience programming in Python, Gearsbot is a spectacular project I love it, they are geniuses if they have a video tutorial or a pdf of how to install the server I would appreciate it

dancito20, 2021-05-13:
https://youtu.be/ILLJ9CovZr0
work of my student: https://photos.google.com/u/1/album/AF1QipOHz9ccgXWlJixlx_KUeQj_IbvEYyeA-TvIkm22/photo/AF1QipNXw9TAWujE0cLXL2ixNNc6xquvBwGORZxJ3CZq

sgmurray, 2021-05-13:
1.  A recommendation that I can give you is that you add the functionality that allows you to move the sensors to the position where it is needed or that we can create our own robot designs

Have you tried using the robot configurator?
https://gears.aposteriori.com.sg/configurator.html
https://github.com/QuirkyCort/gears/wiki/Robot-Configurations

2.   I do not know how difficult it is but if it is at your disposal to be able to import the models made in LEGO DIGITAL DESIGNER OR STUDIO would be a very good update,

This is potentially difficult. Could you give more details why you want this feature? Perhaps you can accomplish most of what you want with the features already included in Gearsbot.

3.  there is an error that generates the program and that is when we want to remove the sensor block when it is joined with the moperator block, the same there generates a problem and the program is damaged

Could you post a screenshot  or attach a file of the code that is broken.

4. distance sensor does not It detects the blocks well, only the edge of the table

Are you talking about the ultrasonic sensor on the cagebot and the blocks on the WRO 2021 junior world? 
Otherwise, could you attach the world with the block you are trying to detect. 

5.  if they have a video tutorial or a pdf of how to install the server I would appreciate it

From the main github page
https://github.com/QuirkyCort/gears

If you do not have a webserver, but have Python3 installed on your computer, you can try...

Download Gears from https://github.com/QuirkyCort/gears/archive/refs/heads/master.zip
Change to the "gears/public" directory
Run python -m http.server 1337 Do not close the window with the Python command running.
This should get the site running on http://localhost:1337 (...try http://127.0.0.1:1337 if that doesn't work).

QuirkyCort, 2021-05-14:
The ultrasonic sensor works by generating a number of rays and detecting if they hit any objects. You can find a picture of the rays here...

https://github.com/QuirkyCort/gears/issues/60#issuecomment-812518810

If the object is small, it may fall between the rays and won't get detected. A real ultrasonic isn't great at detecting small objects either, but the conditions for it to miss and object is different, so you can't directly compare them.

Also, the green blocks in WRO 2021 Junior is below the center of the ultrasonic sensor of the cagebot. If the robot is very near, it'll miss the block as the rays will end up above the block. Again, not unlike a real ultrasonic, although a real ultrasonic will be a little more forgiving in this particular situation. If this is an issue for you, try configuring the robot and placing the sensor lower.

closed 2021-08-12 by QuirkyCort