QuirkyCort/gears #87: Radio Issues

sgmurray, 2021-06-20:
I am having 2 issues with the radio.

1. In single player mode calling radio.send() causes the user code to stop executing with no explanation.  I know it might seem strange to call radio.send() in single player mode, but I often like to test multiplayer code first in single player mode before going through the hassle of exporting as a zip file.  

2. I am getting NotImplemented when trying to read elements of  lists sent by the radio. I have attached 2 zip files to demonstrate. The receiver and sender programs are designed to be run in Arena mode by 2 robots on the same team.  

[reciever (10).zip](https://github.com/QuirkyCort/gears/files/6681977/reciever.10.zip)

[sender (4).zip](https://github.com/QuirkyCort/gears/files/6681979/sender.4.zip)

QuirkyCort, 2021-06-20:
Thanks for the report.

The first issue is because you are sending to the team, but there are no teammates defined when in single player, causing an error to be thrown. I've added in a check, to prevent this.

The second issue because Python list were not correctly mapped when converting to Js. I've fixed that too.

Let me know if you still have problems with it.

sgmurray, 2021-06-20:
The fixes work for me for live version of the site. Thanks.

When you get a chance, it would be nice to have the fixes in the github repository too.

QuirkyCort, 2021-06-21:
Forgot to push. It's in the repository now.

closed 2021-06-21 by sgmurray