QuirkyCort/gears #90: Feature/Pull Request: More Wheels

sgmurray, 2021-07-23:
I recently submitted a pull request for a feature that allows users to add more wheels to robots in GEARS. 
Here is a video demonstrating what is possible with this new feature.
https://youtu.be/Mq4urBJPNng

QuirkyCort, 2021-07-27:
Looks great, but it's gonna take me a while before I can find the time to look through the code. As well, there is a competition going on using Gears, so better to avoid any big changes til it's over.

sgmurray, 2021-07-31:
The stair climber robot in my video is based on the real life EV3 stair climber. 
https://www.youtube.com/watch?v=f8xkVyMFrog

I have added 2 new features to my fork of gears, wheel blocks and passive wheels, which allow the simulated climber to use only 3 motors (just like the physical robot). 

Here's how you can try out the stair climber for yourself. 

1. Go to https://sgmurray.github.io/gears/public/
2. Download and extract [stairClimber.zip](https://github.com/QuirkyCort/gears/files/6911891/stairClimber.zip)
2. Robot > Load from file>stairClimber.json
3. Worlds > Load from file>stairWorld.json
4. File > Load Python from your computer > stairClimberCode.py

QuirkyCort, 2021-08-12:
Sorry for taking a long time. Finally gotten some free time to look through the PR. It's great. Thanks!

sgmurray, 2021-08-21:
@Cort Would you also be willing to add wheel blocks (dumb blocks that look like wheels)? I used them to make linked left and right pairs of wheels on the stair climber. 

https://github.com/sgmurray/gears/commit/a5633079b956af24fe439e1fac7b170119f33141

QuirkyCort, 2021-08-21:
I've already done that. I've named it "WheelPassive" and it's under the "Others" category in the configurator.

There are also a few fixes related to the "more wheels" PR, so you might want to pull that into your own version.

sgmurray, 2021-08-21:
Maybe I'm missing something. WheelPassive creates a hinge joint between the parent and the wheel. What I am asking for an additional component that acts like a dumb block (no hinge joint). On the real stair climber left and right pairs of wheels are attached to the same axle and rotate in lockstep. The way I replicated this in gears is to create a WheelActuator and add a WheelBlock as a child. 

The commit that I linked to does the above. I will make a pull request so it is clearer.

[image]

QuirkyCort, 2021-08-21:
Ok. I think I understand. The WheelBlock is basically just a cylinder, but with the default wheel skin and the magical tire downward force. Anything other differences from a cylinder?

sgmurray, 2021-08-21:
Correct. The wheelBlock is just a cylinder with the wheel skin. Maybe make magical tire downward force a checkbox option that users can choose to turn on/off. I can't think of any other differences from a cylinder.

QuirkyCort, 2021-08-24:
I added in the option on my local copy, then realized that this won't work (...ammo throws an error). The cylinder doesn't have it's own physics body as it is merged with the physics body of the wheel.

The closest I can do is to apply the downward force on the parent (wheel), but at the location of the cylinder. Is that how you did it? That will work, but the center of mass will remain at the wheel, so I don't think that's a great solution. There may also be some minor physics issue due to the position of the joint between the wheel and the body being off to one side.

A better solution may be to place the wheel in the middle and set it to a small diameter, then attach two cylinders to act as the wheels. The total downward force will be only half that of using two separate wheels, but that's easily resolved by making the tire down force a configurable option.

closed 2025-05-29 by QuirkyCort