QuirkyCort/gears #93: Loading worlds and editing Worlds in the World Configurator not working

mrfabroa, 2021-09-24:
I tried loading a world file into the World Builder and the default world is not being updated.  Edits I've tried are selecting a new ground and changing the height of the walls.  The world in the configurator is not updated with these changes.  When trying to load a world the world is still not being updated.  Here are the contents of the World file.

```
{
  "worldName": "grid",
  "options": {
    "imageURL": "",
    "groundType": "box",
    "length": 400,
    "width": 400,
    "uScale": 20,
    "vScale": 20,
    "imageScale": "2.353",
    "timer": "none",
    "timerDuration": 60,
    "timerEnd": "continue",
    "wall": true,
    "wallHeight": 10,
    "wallThickness": 5,
    "wallColor": "B3B3B3",
    "groundFriction": 1,
    "wallFriction": 0.1,
    "groundRestitution": 0,
    "wallRestitution": 0.1,
    "restartAnimationOnRun": false,
    "objects": [],
    "startPos": "center",
    "startPosXYZStr": "",
    "startRotStr": "",
    "startPosXYZ": null,
    "startRot": null,
    "arenaStartPosXYZ": null,
    "arenaStartRot": null,
    "image": "textures/maps/grid.png"
  }
}
```

QuirkyCort, 2021-09-24:
That world file is for a "grid" world. The world builder can only edit "custom" world. I've added some checks in the world builder so that it'll give the appropriate error message now instead of failing quietly.

In your case, the grid world is fortunately very similar to the custom world, so it's possible to convert the file with a few changes...

```
{
  "worldName": "custom",
  "options": {
    "imageURL": "textures/maps/grid.png",
    "groundType": "box",
    "length": 400,
    "width": 400,
    "uScale": 20,
    "vScale": 20,
    "imageScale": "2.353",
    "timer": "none",
    "timerDuration": 60,
    "timerEnd": "continue",
    "wall": true,
    "wallHeight": 10,
    "wallThickness": 5,
    "wallColor": "B3B3B3",
    "groundFriction": 1,
    "wallFriction": 0.1,
    "groundRestitution": 0,
    "wallRestitution": 0.1,
    "restartAnimationOnRun": false,
    "objects": [],
    "startPos": "center",
    "startPosXYZStr": "",
    "startRotStr": "",
    "startPosXYZ": null,
    "startRot": null,
    "arenaStartPosXYZ": null,
    "arenaStartRot": null
  }
}
```

For those who are interested, the technical reasons behind this are...

1. Each world (eg. grid, line following, football) is handled by their own javascript program.
2. The javascript program for each world will have its own set of options that the other world will not understand. For example, the maze world have an option for columns and rows. These options are not meaningful or understood by any other worlds.
3. The javascript program for the "custom" world is designed to allow a high level of customization. This is the only world used by the world builder.
4. The world builder cannot edit other types of world files (eg. maze world), since it doesn't understand how to generate the random maze. As well, these other worlds doesn't know how to load the objects and models available in the world builder.

mrfabroa, 2021-09-24:
Got it, thank you for the adjustments and a very thorough response.

closed 2021-09-24 by mrfabroa