QuirkyCort/gears #110: importing models via URL

mrfabroa, 2022-01-10:
I am trying to import a model file hosted on github, however, when I set the model URL, I get the error(!) object.  I don't think it is the .glb.  For instance, when I try this url hosted on gears it works
https://gears.aposteriori.com.sg/models/Kenny.nl%20-%20Cars/ambulance.glb

However, if I host it on github, https://github.com/SACHSTech/GearsbotEV3Curriculum/blob/main/Models/ambulance.glb it doesn't work.  What else do I need to set on the github end for it to work?

QuirkyCort, 2022-01-10:
That github link doesn't point to the actual ambulance.glb file, it's a link to the github page for the file.

The link to the file should be: https://raw.githubusercontent.com/SACHSTech/GearsbotEV3Curriculum/main/Models/ambulance.glb

Annoyingly, for binary files, github don't display the above link (...for text files, you can get the above link by clicking "Raw"). Instead, github provides a "Download" button and a "View Raw" link, neither which are CORS enabled (...and thus will not work). Still, it's not hard to construct the above link manually.

An alternative is to publish the repository to github pages, and get the link from there (...files served from github pages are CORS enabled).

mrfabroa, 2022-01-10:
Got it, thank you!!

closed 2022-01-10 by mrfabroa