QuirkyCort/gears #116: A "computer" option in Radio for receiver/emitter

DJuego, 2022-05-26:
It's been a long time, 

You should know that we are putting your spectacular project to good use. Huge thanks for everything!

I appreciate your efforts in supporting Pybricks. Although we work under ev3dev and will continue to do so for the time being (we think it suits our needs better), I would like to experiment a bit with it this summer. It is unavoidable now that the EV3 brick has been discontinued. 

In any case; Now, I would like to ask you a question, could a new "computer" or so option be added to the radio combo?

We are working on a challenge where the robotic units will communicate with a server, which centralizes the information and provides the robots with "advanced" services via radio. 

My intention is to implement a class in Python that mimics the operation of your radio blocks, and that works under ev3dev (a Radio Class).

I am not asking you to support "computers or servers" in the simulation (although that would be absolutely great). I understand the added complexity. I would just like to have the possibility to indicate a receiver/emitter that is not a Robot ID. 

For now; as in our specific challenge the robots will not communicate with each other I will use 0 as ID, but naturally, it would be a problem, if in addition to the server, the robots would send messages to each other. 

Thanks again, Hero of Computacional Thinking! :-)

DJuego

DJuego, 2022-05-26:
Well. Thinking about it (and with the desire for greater flexibility.), maybe the best way is to pass as literal text the recipient of the message, as it is done with the name of the mailbox, or something like that. Some words like "all" (broadcast) could be reserved... 

You may prefer to create an alternative more generic block to the existing "Send message to" with this functionality (send message to [TEXT] via mailbox [TEXT])... ?

DJuego

QuirkyCort, 2022-06-23:
Sorry for the late reply. Been busy lately.

The current Radio in Gears is really only meant for the arena, and is not well suited for real robots. Most notably, there are no ways for the robot to declare its ID; It's hard coded based on the position they are loaded in the arena.

I have a nethelper library (meant for simple games networking) which probably runs on ev3dev (...not tested) and would be better for real network communications, but it's a little harder to use and doesn't support sending to only a specific group.

I would think that for a real Radio class, we'll need to change from...

```
radio = Radio()
radio.send(0, "default", some_value)
```

...to...

```
radio = Radio("server address", "my ID")
radio.send("dest ID", "default", some_value)
radio.send(["ID1", "ID2"], "default", some_value)
```

While not particularly complicated, this is unnecessary for the arena where there is always a max of 4 robots and the teams are fixed. But if there is a real robot Radio library, I'll follow the real library so that the code in Gears can be compatible with real robots.

Let me know if you are working on a Radio class that works with a physical robot.

achiyae, 2022-09-14:
Hi, 

I have a similar problem with a different use case. I am a software-engineering researcher, and I use EV3 to evaluate new programming paradigms that I develop. To simplify the research, I wish to use gears, but my programming languages are neither JS nor Python. Therefore, I want to create a "relay" robot that listens for requests over TCP/IP and sends the results back. These requests can be, e.g., retrieving sensor readings, rotating a motor, etc. 

As for known radion libraries - I think that Mosquitto and ROS are both great options.

@DJuego Did you make any progress?

Thanks, Achiya