QuirkyCort/gears #122: Support a custom robot

vanminh0910, 2023-01-10:
Hi, this is great simulator tool for robot. We have tried and used in many of our classes.

My question is how to add support for a custom robot (running micropython with esp32)?

QuirkyCort, 2023-01-11:
You have a custom robot running micropython on esp32? If you could share more, I would be very interested to hear about it. I've been mulling over designing an opensource esp32 based robotics platform as a physical counterpart to gearsbot for a while now.

If you want to add your own robot api into gearsbot, you'll need to...

- Add the python libraries (see "public/pybricks" for an example)
- When writing the python libraries, you can use the functions provided in "public/js/simPython.js" to control the robot.
- Tell Skulpt about the libraries ("public/js/skulpt" line 4)
- Add a generator (see "public/js/pybricks_generator" for an example). This is only needed if you need to convert the user's block program to python. If the user is programing gearsbot directly in python, this is not required.
- Edit main.js to add a menu entry for your new generator.

vanminh0910, 2023-01-11:
Hi,

That sounds great. I will check and try and get back if any issue.

We have 2 Python robots for education using ESP32. You can find them here https://ohstem.vn/xbot/ and https://ohstem.vn/product/robot-stem-rover/.

If you have any interest, we can send you a few to try. We can share the design and Python API source if you need. They are mainly for STEM education so we only want to do all the best to support the kids. A great simulator tool like Gearsbot is perfect for use.

QuirkyCort, 2023-01-12:
Thanks for the offer! Won't need you to send me the robot, but if you have code to share, that would be appreciated.

If you have any questions about gearsbot, feel free to ask here.

closed 2025-05-29 by QuirkyCort